intel/media-driver #1505: [Bug]: failure to build ToT or 22.4.4

satmandu, 2022-10-03:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

1. on Chromebrew x86_64, with libva 0.16, gmmlib 22.2.0, build fails with this message.
```
[497/1177] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9_bxt/codec/hal/codechal_decode_nv12top010_kernel_g9_bxt.cpp.o
[498/1177] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp.o
FAILED: media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp.o 
/usr/local/lib64/ccache/bin/c++ -DCLASS_TRACE=0 -DENABLE_KERNELS -DHEVC_FEI_ENABLE_CMRT -DIGFX_DG2_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLN_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_DG1_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DIGFX_XEHP_SDV_CMFCPATCH_SUPPORTED -DIGFX_XEHP_SDV_SUPPORTED -DIGFX_XE_HPG_CMFCPATCH_SUPPORTED -DIGFX_XE_HPG_SUPPORTED -DMEDIA_VERSION=\"22.6.0\" -DMEDIA_VERSION_DETAILS=\"c7132d8\" -DVEBOX_AUTO_DENOISE_SUPPORTED=1 -DX11_FOUND -D_AV1_DECODE_SUPPORTED -D_AV1_ENCODE_VDENC_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_AVC_ENCODE_VME_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_DECODE_PROCESSING_SUPPORTED -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_HEVC_ENCODE_VME_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_MPEG2_ENCODE_VME_SUPPORTED -D_RELEASE -D_VC1_DECODE_SUPPORTED -D_VP8_DECODE_SUPPORTED -D_VP8_ENCODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/linux/common/os/i915/include -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/linux/common/os/i915/include/uapi -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/os -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/linux/common/os -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_common/agnostic/common/os -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_common/linux/common/os -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_softlet/agnostic/common/os -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_softlet/linux/common/os/osservice -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_softlet/linux/common/os -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/cm -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/media_interfaces -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/shared/user_setting -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/shared/mediacopy -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/common/shared -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen8/cm -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen8/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen8/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen8/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen8_bdw/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen8_bdw/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9/cm -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9_bxt/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9_skl/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9_glk/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9_kbl/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen10/cm -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen10/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen10/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen10/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen11/cm -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen11/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen11/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen11/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen11_jsl_ehl/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen12/cm -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen12/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen12/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen12/shared -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/g12/g12_base/hw/render -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_M/Xe_M_base/shared -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_M/Xe_XPM/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_M/Xe_XPM/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_M/Xe_HPM/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_M/Xe_HPM/hw -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_R/Xe_HP_base/hw/render -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_R/Xe_HP_base/hw/blt -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_R/Xe_HP/hw/render -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/Xe_R/Xe_HPG/hw/render -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/common/shared/media_sfc_interface -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/common/shared/scalability -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/m12/m12_0/shared/mediacopy -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/g12/g12_base/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/g12/g12_0/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/g12/g12_1/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/Xe_R/Xe_HP_Base/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/media_softlet/agnostic/Xe_R/Xe_HP/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_ [... cut ...]
/usr/local/tmp/crew/intel_media_driver.20221003201511.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2915:16: note: object ‘CodechalEncHevcStateG9::m_encBTu7BCurbeInit’ of size 224
 2915 | const uint32_t CodechalEncHevcStateG9::m_encBTu7BCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

### What's the usage scenario when you are seeing the problem?

Others

### Debug Information

libva 0.16, gmmlib 22.2.0

build log:
[intel_media_driver-22.6.0-c7132d8-x86_64.log](https://github.com/intel/media-driver/files/9700894/intel_media_driver-22.6.0-c7132d8-x86_64.log)

XinfengZhang, 2022-10-08:
which version is good? these code is gen9, I suppose it already be stable long time.

stellawuintel, 2022-10-08:
Auto Created VSMGWL-57546 for further analysis.

satmandu, 2022-10-08:
> which version is good? these code is gen9, I suppose it already be stable long time.

The last build we did was for 21.2.3.

XinfengZhang, 2022-10-09:
how about the build env? such as gcc version, cmake version etc ?

satmandu, 2022-10-10:
Build is in a docker container.

```
cmake --version
cmake version 3.24.2
gcc --version
gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Commands to replicate our build environment:

```
#!/bin/bash
docker pull satmandu/crewbuild:nocturne-x86_64.m90
docker run --platform linux/amd64 --rm -v $(pwd)/pkg_cache:/usr/local/tmp/packages -v $(pwd):/output -h $(hostname)-x86_64 -it satmandu/crewbuild:nocturne-x86_64.m90 /usr/local/bin/setarch x86_64 sudo -i -u chronos /usr/local/bin/bash -i
```
Followed by running this to get current packages inside the container:
```
CREW_TESTING_REPO=https://github.com/satmandu/chromebrew.git CREW_TESTING_BRANCH=gnome43 CREW_TESTING=1 crew update ; crew upgrade
```
To get the package file we are trying to build one would do this afterwards in the container:

```
cat << 'PACKAGEEOF' > intel_media_driver.rb
require 'package'

class Intel_media_driver < Package
  description 'The Intel(R) Media Driver for VAAPI is a new VA-API (Video Acceleration API) user mode driver supporting hardware accelerated decoding, encoding, and video post processing for GEN based graphics hardware.'
  homepage 'https://github.com/intel/media-driver'
  @_ver = '22.6.0-c7132d8'
  version @_ver
  license 'BSD-3, and MIT'
  compatibility 'x86_64'
  source_url 'https://github.com/intel/media-driver.git'
  git_hashtag "c7132d869bed4d9748b789ed4aeb06cc49a311d5"

  depends_on 'gmmlib'
  depends_on 'libva'

  def self.preflight
    abort 'Not an Intel processor, aborting.'.lightred unless `grep -c 'GenuineIntel' /proc/cpuinfo`.to_i.positive?
  end

  def self.build
    FileUtils.mkdir('builddir')
    Dir.chdir('builddir') do
      system "cmake #{CREW_CMAKE_OPTIONS.sub('-pipe', '-pipe -Wno-error')} ../ -G Ninja"
    end
    system 'ninja -C builddir'
  end

  def self.install
    system "DESTDIR=#{CREW_DEST_DIR} ninja -C builddir install"

    FileUtils.mkdir_p "#{CREW_DEST_PREFIX}/etc/env.d/"
    @env = <<~EOF
      # intel_media_driver configuration
      export LIBVA_DRIVER_NAME=iHD
    EOF
    File.write("#{CREW_DEST_PREFIX}/etc/env.d/intel_media_driver", @env)
  end
end
PACKAGEEOF
```

To generate the build we would run `crew build -k intel_media_driver` and then the build dir would end up in `/usr/local/tmp/crew/` .

(The full build log from such a build is attached in the original post above.)

satmandu, 2022-10-10:
>
> Would you please help update libva to 1.17.0 and then try again? The issue couldn't be reproduced in my environment after I upgrade libva to 1.17.0

Which git commit should I build? Anything recent? I see 1.17.0 has not been tagged yet.

cqian2, 2022-10-11:
Can you try git commit 65c67dd2bd6b68afa5d86a8f9bdba6aa592e686f ?

satmandu, 2022-10-11:
Thanks! That fixes the build!

satmandu, 2022-10-11:
False report! The build failed.

```
[501/1177] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp.o
FAILED: media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp.o 
/usr/local/bin/c++ -DCLASS_TRACE=0 -DENABLE_KERNELS -DHEVC_FEI_ENABLE_CMRT -DIGFX_DG2_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLN_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_DG1_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DIGFX_XEHP_SDV_CMFCPATCH_SUPPORTED -DIGFX_XEHP_SDV_SUPPORTED -DIGFX_XE_HPG_CMFCPATCH_SUPPORTED -DIGFX_XE_HPG_SUPPORTED -DMEDIA_VERSION=\"22.6.0\" -DMEDIA_VERSION_DETAILS=\"c7132d8\" -DVEBOX_AUTO_DENOISE_SUPPORTED=1 -DX11_FOUND -D_AV1_DECODE_SUPPORTED -D_AV1_ENCODE_VDENC_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_AVC_ENCODE_VME_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_DECODE_PROCESSING_SUPPORTED -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_HEVC_ENCODE_VME_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_MPEG2_ENCODE_VME_SUPPORTED -D_RELEASE -D_VC1_DECODE_SUPPORTED -D_VP8_DECODE_SUPPORTED -D_VP8_ENCODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/linux/common/os/i915/include -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/linux/common/os/i915/include/uapi -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/os -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/linux/common/os -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_common/agnostic/common/os -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_common/linux/common/os -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_softlet/agnostic/common/os -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_softlet/linux/common/os/osservice -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_softlet/linux/common/os -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/cm -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/hw -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/media_interfaces -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/shared/user_setting -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/shared/mediacopy -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/common/shared -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen8/cm -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen8/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen8/hw -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen8/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen8_bdw/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen8_bdw/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9/cm -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9/hw -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9_bxt/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9_skl/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9_glk/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9_kbl/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen10/cm -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen10/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen10/hw -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen10/renderhal -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen11/cm -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen11/hw/vdbox -I/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen11/hw [... cut ...]
/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2904:16: note: object ‘CodechalEncHevcStateG9::m_encBTu4BCurbeInit’ of size 224
 2904 | const uint32_t CodechalEncHevcStateG9::m_encBTu4BCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/tmp/crew/intel_media_driver.20221011143424.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2915:16: note: object ‘CodechalEncHevcStateG9::m_encBTu7BCurbeInit’ of size 224
 2915 | const uint32_t CodechalEncHevcStateG9::m_encBTu7BCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

satmandu, 2022-10-11:
Build log:

[intel_media_driver-22.6.0-c7132d8-x86_64.log](https://github.com/intel/media-driver/files/9757517/intel_media_driver-22.6.0-c7132d8-x86_64.log)

cqian2, 2022-10-13:
I double compared your configs w/ mine but couldn't find any diff.
The code where the error happened has been frozen for a long time and it's not possible for us to change the code now. 
Hense, for a work around, would you please just disable the "all warnings being treated as errors" though below two ways, and then try again?
1. Remove line-258 "set_target_properties(${LIB_NAME}_CODEC PROPERTIES COMPILE_FLAGS "-Werror")" from media_driver/media_top_cmake.cmake
or
2. Add "-DMEDIA_BUILD_FATAL_WARNINGS=OFF" in your cmake command

satmandu, 2022-10-13:
Thanks for that. Looks like option number 2 lets me get past that error:

```
[502/1177] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9_bxt/codec/hal/codechal_vdenc_avc_g9_bxt.cpp.o
[503/1177] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp.o
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp: In member function ‘CodechalEncHevcStateG9::SetupBrcConstantTable(MOS_SURFACE*)’:
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:6193:23: warning: array subscript ‘struct CODECHAL_ENC_HEVC_B_MB_ENC_CURBE_G9[0]’ is partly outside array bounds of ‘uint32_t[56]’ [-Warray-bounds]
 6193 |         if(curbe->DW3.BlockBasedSkipEnable)
      |                       ^~~~~~~~~~~~~~~~~~~~
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2950:16: note: object ‘CodechalEncHevcStateG9::m_encBTu7PCurbeInit’ of size 224
 2950 | const uint32_t CodechalEncHevcStateG9::m_encBTu7PCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2893:16: note: object ‘CodechalEncHevcStateG9::m_encBTu1BCurbeInit’ of size 224
 2893 | const uint32_t CodechalEncHevcStateG9::m_encBTu1BCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2938:16: note: object ‘CodechalEncHevcStateG9::m_encBTu4PCurbeInit’ of size 224
 2938 | const uint32_t CodechalEncHevcStateG9::m_encBTu4PCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2926:16: note: object ‘CodechalEncHevcStateG9::m_encBTu1PCurbeInit’ of size 224
 2926 | const uint32_t CodechalEncHevcStateG9::m_encBTu1PCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2904:16: note: object ‘CodechalEncHevcStateG9::m_encBTu4BCurbeInit’ of size 224
 2904 | const uint32_t CodechalEncHevcStateG9::m_encBTu4BCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/tmp/crew/intel_media_driver.20221013223025.dir/media_driver/agnostic/gen9/codec/hal/codechal_encode_hevc_g9.cpp:2915:16: note: object ‘CodechalEncHevcStateG9::m_encBTu7BCurbeInit’ of size 224
 2915 | const uint32_t CodechalEncHevcStateG9::m_encBTu7BCurbeInit[56] =
      |                ^~~~~~~~~~~~~~~~~~~~~~
[504/1177] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9_skl/codec/hal/codechal_fei_avc_g9_skl.cpp.o
[505/1177] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen9_skl/codec/hal/codechal_encode_mpeg2_g9_skl.cp
```

cqian2, 2022-10-14:
That's great. 
Since we probably will not change the frozen code, thus please use this workaround unless there's a must to solve this compile error. Please help close this issue.

satmandu, 2022-10-14:
And compilation succeeded. Thanks again!