intel/media-driver #1506: [Bug]:[TGL-U][ffmpeg-vaapi] GPU hang when decoding HEVC video with ffmpeg and HCP scalability enabled.

UngTengEn, 2022-10-06:
### Which component impacted?

Decode

### What happened?

1. TGL-U i5-1145G7E; ubuntu 22.05 and kernel 5.15.0-48-generic.
2. Install or build libva 2.15.0 and media-driver 22.4.4 with HCP Scalability support.
3. Install ffmpeg 4.4.2
4. Download this video https://images-assets.nasa.gov/video/First-8K-Video-from-Space/First-8K-Video-from-Space~orig.mp4
5. With HCP Scalibility enabled, decode with ffmpeg-vaapi : "ffmpeg -loglevel verbose -hide_banner -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi_vld -stream_loop 20 -i /storage/sd/nasa.mp4 -f null -".
6. Results in a GPU hang after some time, typically within about 10 minutes of runtime (but sometimes it does successfully get through all 20 loop iterations).
7. igfx_user_feature.txt contents :- cat /etc/igfx_user_feature.txt 
[KEY]
        0x00000001
        UFKEY_INTERNAL\LibVa
                [VALUE]
                        Disable MMC
                        4
                        1
                [VALUE]
                        Enable HCP Scalability Decode
                        4
                        1

### What's the usage scenario when you are seeing the problem?

Video Analytics

### Debug Information

1. Please check the vainfo.
2.root@p12dltglu208-Tiger-Lake-Client-Platform:~/Videos# ls /dev/dri
by-path  card0  renderD128
[vainfo.log](https://github.com/intel/media-driver/files/9721008/vainfo.log)
[dmesg.log](https://github.com/intel/media-driver/files/9721009/dmesg.log)

stellawuintel, 2022-10-10:
Auto Created VSMGWL-57651 for further analysis.

Jexu, 2022-10-11:
The large resolution decode may trigger vcs preemption timeout.
Can you try to enlarge heartbeat interval and preemption timeout by bellow cmd.

for i in /sys/class/drm/card0/engine/vcs*/heartbeat_interval_ms; do echo 3500 > $i; done;
for i in /sys/class/drm/card0/engine/vcs*/preempt_timeout_ms; do echo 2640 > $i; done;

UngTengEn, 2022-10-12:
Hi, try out above settings but the GPU Hang still happenings.

_root@p12dltglu208-Tiger-Lake-Client-Platform:# for i in /sys/class/drm/card0/engine/vcs*/heartbeat_interval_ms; do echo $i; cat $i; done;
/sys/class/drm/card0/engine/vcs0/heartbeat_interval_ms
3500
/sys/class/drm/card0/engine/vcs1/heartbeat_interval_ms
3500
root@p12dltglu208-Tiger-Lake-Client-Platform:# for i in /sys/class/drm/card0/engine/vcs*/preempt_timeout_ms; do echo  $i; cat $i; done;
/sys/class/drm/card0/engine/vcs0/preempt_timeout_ms
2640
/sys/class/drm/card0/engine/vcs1/preempt_timeout_ms
2640
root@p12dltglu208-Tiger-Lake-Client-Platform:# dmesg 
[20490.830033] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
[20490.830061] i915 0000:00:02.0: [drm] ffmpeg[7094] context reset due to GPU hang
[20490.968490] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:4:28fffffd, in ffmpeg [7094]
[20500.072533] Fence expiration time out i915-0000:00:02.0:ffmpeg[7094]:8ca4!
[20500.079063] Fence expiration time out i915-0000:00:02.0:ffmpeg[7094]:8ca8!_

You know what is the maximum value we can set for those settings?

Jexu, 2022-10-13:
Could you help try above patch? I just ran 100 times with this fix, didn't see hang anymore.

UngTengEn, 2022-10-17:
I tried that patch it seems fix the problem.