intel/media-driver #1507: [Bug]: Use of uninitialized memory.

stolk, 2022-10-06:
### Which component impacted?

Decode, Encode, Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

The driver will access unintialized memory, leading to undefined behaviour.

This error can be caught with running valgrind.

```
$ valgrind ./vainfo 
==33231== Memcheck, a memory error detector
==33231== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==33231== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==33231== Command: ./vainfo
==33231== 
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
==33231== Conditional jump or move depends on uninitialised value(s)
==33231==    at 0x484A830: rindex (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==33231==    by 0x52B4C49: parse_separate_sysfs_files(int, int, _drmPciDeviceInfo*, bool) (drm_device.h:596)
==33231==    by 0x52B520D: drmParsePciDeviceInfo(int, int, _drmPciDeviceInfo*, unsigned int) (drm_device.h:699)
==33231==    by 0x52B5573: drmProcessPciDevice(_drmDevice**, char const*, int, int, int, bool, unsigned int) (drm_device.h:823)
==33231==    by 0x52B5FCC: process_device(_drmDevice**, char const*, int, bool, unsigned int) (drm_device.h:1079)
==33231==    by 0x52B6401: drmGetDevice2(int, unsigned int, _drmDevice**) (drm_device.h:1252)
==33231==    by 0x52B65C6: drmGetDevice(int, _drmDevice**) (drm_device.h:1298)
==33231==    by 0x52B9EC6: MosInterface::GetAdapterBDF(_MOS_OS_CONTEXT*, _ADAPTER_BDF_*) (mos_interface.cpp:613)
==33231==    by 0x5305EEB: OsContextSpecificNext::Init(void*) (mos_context_specific_next.cpp:178)
==33231==    by 0x52B6E39: MosInterface::CreateOsDeviceContext(void*, OsContextNext**) (mos_interface.cpp:127)
==33231==    by 0x5888E44: DdiMedia_InitMediaContext(VADriverContext*, int, int*, int*, bool&) (media_libva.cpp:1791)
==33231==    by 0x58889EF: DdiMedia__Initialize(VADriverContext*, int*, int*) (media_libva.cpp:1693)
==33231== 
==33231== Conditional jump or move depends on uninitialised value(s)
==33231==    at 0x4A78505: __vfprintf_internal (vfprintf-process-arg.c:368)
==33231==    by 0x4A9BD8B: __vsnprintf_internal (vsnprintf.c:114)
==33231==    by 0x4A745D5: snprintf (snprintf.c:31)
==33231==    by 0x52B4C9A: parse_separate_sysfs_files(int, int, _drmPciDeviceInfo*, bool) (drm_device.h:598)
==33231==    by 0x52B520D: drmParsePciDeviceInfo(int, int, _drmPciDeviceInfo*, unsigned int) (drm_device.h:699)
==33231==    by 0x52B5573: drmProcessPciDevice(_drmDevice**, char const*, int, int, int, bool, unsigned int) (drm_device.h:823)
==33231==    by 0x52B5FCC: process_device(_drmDevice**, char const*, int, bool, unsigned int) (drm_device.h:1079)
==33231==    by 0x52B6401: drmGetDevice2(int, unsigned int, _drmDevice**) (drm_device.h:1252)
==33231==    by 0x52B65C6: drmGetDevice(int, _drmDevice**) (drm_device.h:1298)
==33231==    by 0x52B9EC6: MosInterface::GetAdapterBDF(_MOS_OS_CONTEXT*, _ADAPTER_BDF_*) (mos_interface.cpp:613)
==33231==    by 0x5305EEB: OsContextSpecificNext::Init(void*) (mos_context_specific_next.cpp:178)
==33231==    by 0x52B6E39: MosInterface::CreateOsDeviceContext(void*, OsContextNext**) (mos_interface.cpp:127)
==33231== 
==33231== Use of uninitialised value of size 8
==33231==    at 0x484AD16: strlen (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==33231==    by 0x4A79E94: __vfprintf_internal (vfprintf-process-arg.c:397)
==33231==    by 0x4A9BD8B: __vsnprintf_internal (vsnprintf.c:114)
==33231==    by 0x4A745D5: snprintf (snprintf.c:31)
==33231==    by 0x52B4C9A: parse_separate_sysfs_files(int, int, _drmPciDeviceInfo*, bool) (drm_device.h:598)
==33231==    by 0x52B520D: drmParsePciDeviceInfo(int, int, _drmPciDeviceInfo*, unsigned int) (drm_device.h:699)
==33231==    by 0x52B5573: drmProcessPciDevice(_drmDevice**, char const*, int, int, int, bool, unsigned int) (drm_device.h:823)
==33231==    by 0x52B5FCC: process_device(_drmDevice**, char const*, int, bool, unsigned int) (drm_device.h:1079)
==33231==    by 0x52B6401: drmGetDevice2(int, unsigned int, _drmDevice**) (drm_device.h:1252)
==33231==    by 0x52B65C6: drmGetDevice(int, _drmDevice**) (drm_device.h:1298)
==33231==    by 0x52B9EC6: MosInterface::GetAdapterBDF(_MOS_OS_CONTEXT*, _ADAPTER_BDF_*) (mos_interface.cpp:613)
==33231==    by 0x5305EEB: OsContextSpecificNext::Init(void*) (mos_context_specific_next.cpp:178)
==33231== 
==33231== Use of uninitialised value of size 8
==33231==    at 0x484AD24: strlen (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
[... 140 lines cut ...]
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointEncSlice
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

XinfengZhang, 2022-10-07:
hi @stolk , the patch you attached looks good, but it is still not a PR

stolk, 2022-10-07:
> hi @stolk , the patch you attached looks good, but it is still not a PR

Thanks, @XinfengZhang ,

I'm sorry. Did I forget a step? I think it's PR 1508?
Let me know if I need to do something else.

https://github.com/intel/media-driver/pull/1508

XinfengZhang, 2022-10-08:
sorry,  previous link just show the commit in your folk repo.

stolk, 2022-10-12:
I've created a new pull request for this: #1511

Jexu, 2022-11-24:
There is a comment in the PR, please help update it and then we can get it merged to close this issue. Thanks.

stolk, 2022-11-25:
That comment was invalid, and I replied to it earlier, explaining why the comment was incorrect.

Jexu, 2022-12-02:
Thanks for the fix and pr has been merged. So I will close this issue and please free to re-open it again if having any other questions.

closed 2022-12-02 by Jexu

james-tang17, 2023-06-05:
> That comment was invalid, and I replied to it earlier, explaining why the comment was incorrect.

Hi stolk, did some experiment on my system.

char array[100] = ""; 

can initialize all the member of this array into 0, not only the first one. My gcc version is 9.4.0