intel/media-driver #1513: [Feature]: Performance optimizations?

m911t, 2022-10-13:
### What Feature?

https://www.phoronix.com/news/RadeonSI-Multi-Slice-Encode
https://www.phoronix.com/news/RadeonSI-VCN-Async-Mesa-22.3

According to these articles, there's ways to accelerate vaapi performance. Does i-m-d implement these? I'm quite happy with my intel tigerlake igpu and in the future I'm gonna get a laptop with also intel arc graphics, so this is important to me (and I bet everyone, too.)

### What's the usage scenario would be benifited?

Others

### What impacted?

pretty much, everything, no?

### Do you want to contribute a patch to develop this feature?

No.

Jexu, 2022-11-28:
For second async, it is mainly for up leayer runtime feature how to syncBuffer. If you tried intel sample decoder with one-vpl, it has support with this async option.

eero-t, 2022-12-19:
> If you tried intel sample decoder with one-vpl, it has support with this async option.

Both MFX & VPL transcoding tools support async option, but I do not think they enable it by default.  AFAIK FFmpeg uses by default values corresponding to "-async 4" option for the MFX & VPL "sample_multi_transcode" tool ( e.g. from the "libvpl-tools" package in Intel drivers repository).

OttoYang, 2023-03-28:
We've already supported these features, and users can enable through the parameters in VPL: `NumSlice` and `AsyncDepth`.  
And could we close this issue? @L911P

L911P, 2023-03-28:
yeah, definitely

closed 2023-03-28 by L911P