intel/media-driver #1514: [Bug]: Wrong GPU (1st one) used when STDIN is closed

eero-t, 2022-10-17:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Use-case:
* Run any[1] media application in multi-GPU setup, where 1st GPU is NOT the correct one, and application STDIN is closed

Expected outcome:
* Media driver uses the specified Intel GPU

Actual outcome:
* Wrong (1st) GPU is tried, and program fails

[1] Bug manifests with all applications tested so far:
*  `vainfo` (using VA-API)
* `ffmpeg` (using QSV/MediaSDK)
* `vpl-inspect` & `sample_multi_transcode` (using oneVPL)

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

All use-cases are impacted when application is run under the specified setup (STDIN closed, multiple GPUs, 1st GPU is not the correct one).

### Debug Information

@dvrogozh recommended filing this to media-driver project for better visibility, but this is most likely libva issue, so I have the main bug report there with test program and more information: https://github.com/intel/libva/issues/643

### Do you want to contribute a patch to fix the issue?

No.

eero-t, 2023-01-18:
Potential fix (https://github.com/intel/media-driver/commit/1bf79a0f2da9bf3899e4567510519e0ec7d7d31d) mentioned in https://github.com/intel/libva/issues/643, and included to latest media-driver project release(s), unfortunately did not fix this issue.

When STDIN is closed (= media-driver gets 0 as device FD), things fail even when there's only single GPU, and it's the first (`renderD128`) one.  I.e. this is not just a multi-GPU issue.

XinfengZhang, 2023-05-06:
so, you are suspect that there are some check inside media driver, such as if(fd > 0) ...

eero-t, 2023-05-08:
It's nearly 5 months since my comment.  I don't remember any more what I thought at that moment, or details of how the driver failed, or whether all the indicated test-cases failed (or whether I even tested all of them, I might have tried just one).  But another fd>0 check in the media stack is indeed likely...

Tester is attached to the linked https://github.com/intel/libva/issues/643 issue, so you should be able to easily try the issue with latest Media driver.

This is enough:
```
$ sudo apt install golang vainfo
$ go build vpl-tester.go
$ vpl-tester -close-in  /usr/bin/vainfo --display drm --device /dev/dri/renderD128
```

Works fine with `i965` driver (on GEN9), fails with `iHD` (and works with `iHD` when STDIN is not closed).

You do not need multi-GPU setup, it fails even with single iGPU.

XinfengZhang, 2023-05-08:
thanks @eero-t ,
 hi , @MicroYY please help to check the issue.

eero-t, 2023-05-08:
You do not need multi-dGPU setup, it fails even with single iGPU.

I tried recent media driver build on GEN9 BXT (as that's supported also by previous `i965` driver):
```
$ sudo apt install golang vainfo
$ go build vpl-tester.go
$ ./vpl-tester -close-in  /usr/bin/vainfo --display drm --device /dev/dri/renderD128
```

Above works fine with `i965` media driver, but fails with `iHD`.  I.e. this is also regression from the previous driver.

Default Ubuntu 22.04 `i965` driver:
```
$ ./vpl-tester  -close-in  /usr/bin/vainfo --display drm --device /dev/dri/renderD128
2023/05/08 12:46:18 stdin closed
2023/05/08 12:46:18 stdout/stderr mapped to parent stdout/stderr
2023/05/08 12:46:18 Running: [/usr/bin/vainfo --display drm --device /dev/dri/renderD128]
...
vainfo: VA-API version: 1.19 (libva 2.12.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Broxton - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
....
```

Recent `iHD` driver with STDIN closed:
```
$ export LIBVA_DRIVER_NAME=iHD
$ ./vpl-tester  -close-in  /usr/bin/vainfo --display drm --device /dev/dri/renderD128
2023/05/08 12:47:52 stdin closed
2023/05/08 12:47:52 stdout/stderr mapped to parent stdout/stderr
2023/05/08 12:47:52 Running: [/usr/bin/vainfo --display drm --device /dev/dri/renderD128]
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /opt/install/Nightly/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva error: /opt/install/Nightly/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
2023/05/08 12:47:52 [/usr/bin/vainfo --display drm --device /dev/dri/renderD128] = 3 (0.012412s)
```

And not closed:
```
$ /usr/bin/vainfo --display drm --device /dev/dri/renderD128
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /opt/install/Nightly/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.19 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.2.2 (5dfe5b1d1)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
...
```

MicroYY, 2023-08-18:
Fixed by https://github.com/intel/media-driver/commit/407bd9348 and close this issue.
Pls feel free to reopen if the issue still exist.

closed 2023-08-18 by MicroYY