intel/media-driver #1516: [Feature]: Do we still need the prod KMD on DG2 with linux 6.0 release?

nyanmisaka, 2022-10-18:
### What Feature?

AFAIK DG2(Arc A380) on Ubuntu LTS requires OEM kernel and intel's backported DKMS to work as per [the docs](https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-jammy-arc.html).

I heard that the linux 6.0 release finally has support for DG2. Do we still need the prod KMD on DG2 with linux 6.0 release? If still needed, what pending patches are blocking this?

### What's the usage scenario would be benifited?

Transcode for media delivery

### What impacted?

Feasibility to use media-driver with latest 6.0 kernel release.

Jexu, 2022-11-22:
The latest upstream kernel has supported dg2 and prod KMD is not required.

Jexu, 2022-11-25:
Yeah, this expected change has been in latest drm-tip.

Jexu, 2022-11-28:
Since latest drm-tip has supported dg2 with default probe, so i will close this issue now and please re-open it again if having other questions.

closed 2022-11-28 by Jexu