intel/media-driver #1517: [Bug]: Where is the source of media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_VC1_OLP.c?

sebastinas, 2022-10-19:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

I'm trying to upgrade the free kernel build of the media-driver package in Debian. With the current version, the build requires media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_VC1_OLP.c if built with `-DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF`. I am however unable to find the source for this shader. Please confirm the status of this shader with and add the source if it is a free shader or otherwise skip it if the non-free shaders are disabled.

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

No.

Jexu, 2022-10-20:
Although we upstream this binary, but vc1 decode is defeatured from XE and we don't have source upstream for this binary.

By the way, what you mean it is good in old versino, does you old version code also contain this binary or not?

sebastinas, 2022-10-20:
22.5.2 is behaving correctly, i.e., the free build does not use that shader. A `git bisect` suggests that regression was introduced in fbfe43455c261c99a31ec7d6d8b3d6bb4c664578.

Jexu, 2022-10-20:
@Xiaogangli-intel could you help comment if above commit may invlove this kernel binary compiling.

Xiaogangli-intel, 2022-10-20:
Hi @sebastinas, did you use other flags except `-DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF`? I tried on local, didn't met your issue. And the file is right at [here](https://github.com/intel/media-driver/blob/master/media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_VC1_OLP.c).

dvrogozh, 2022-10-20:
To reproduce the issue, do this:
```
$ git rm media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_VC1_OLP.c
rm 'media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_VC1_OLP.c'

$ cd _build/
$ cmake -DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF ..
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1

<.....>

CMake Error at media_driver/media_top_cmake.cmake:195 (add_library):
  Cannot find source file:

    /home/dvrogozh/git/github/media-driver/media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_VC1_OLP.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  media_driver/CMakeLists.txt:95 (include)

CMake Error at media_driver/media_top_cmake.cmake:195 (add_library):
  No SOURCES given to target: iHD_drv_video_CODEC
Call Stack (most recent call first):
  media_driver/CMakeLists.txt:95 (include)

-- Build files have been written to: /home/dvrogozh/git/github/media-driver/_build
```

PR #1518 (2022-10-21): [Decode] Add missing conditional check for None-Free Kernel build
Fix the issue of missing contitional checking for None-Free kernel build in cmake file. Fixes #1517.
commit 6e25d8e2 (2022-10-21): [Decode] Add missing conditional check for None-Free Kernel build

Fix the issue of missing contitional checking for None-Free kernel build
in cmake file. Fixes #1517.
closed 2022-10-21 by intel-mediadev