intel/media-driver #1519: [Bug]:

heitbaum, 2022-10-23:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Build error with media-driver: update to 2.6.0 and gcc-12.2.0
```
../media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp: In member function 'virtual MOS_STATUS vp::VpRenderHdrKernel::GetCurbeState(void*&, uint32_t&)':
../media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp:3433:11: error: storing the address of local variable 'HDRStatic' in '*curbe' [-Werror=dangling-pointer=]
 3433 |     curbe = (void*)(&HDRStatic);
      |     ~~~~~~^~~~~~~~~~~~~~~~~~~~~
../media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp:2817:34: note: 'HDRStatic' declared here
 2817 |     MEDIA_WALKER_HDR_STATIC_DATA HDRStatic           = g_cInit_MEDIA_STATIC_HDR;
      |                                  ^~~~~~~~~
../media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp:2817:34: note: 'curbe' declared here
cc1plus: all warnings being treated as errors
[1081/1159] Building CXX object media_driver/CMakeFiles/iHD_drv_....dir/__/media_softlet/agnostic/common/vp/hal/utils/vp_visa.cpp.o
ninja: build stopped: subcommand failed.
FAILURE: s/build media-driver during make_target (default)
*********** FAILED COMMAND ***********
ninja ${NINJA_OPTS} ${PKG_MAKE_OPTS_TARGET}
**************************************
docker@d802ff48810a:/var/media/DATA/home-rudi/LibreELEC.tv$ v
```

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Build error

### Do you want to contribute a patch to fix the issue?

No.

cqian2, 2022-10-24:
Hi, 

Would you please help share more config info for this issue?
I tried to build media-driver 22.6.0 w/ gcc 12.2.0 on my env but didn't encounter any issue. Please find my process below:

~/build_media# rm -rf *
~/build_media# cmake -DCMAKE_INSTALL_PREFIX=/usr ../media-driver
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_INSTALL_PREFIX = /usr
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'libva>=1.8.0'
--   Found libva, version 1.17.0
*BUILD_TYPE not defined, default to: release
-- Found X11: /usr/include
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Checking for module 'igdgmm>=12.0.0'
--   Found igdgmm, version 12.2.0
-- media -- PLATFORM = linux
-- media -- ARCH = 64
-- media -- CMAKE_CURRENT_LIST_DIR = /home/intel/Desktop/dev_media/gfx-driver/Source/media-driver/media_driver
-- media -- INCLUDED_LIBS =
-- media -- LIB_NAME = iHD_drv_video
-- media -- OUTPUT_NAME =
-- media -- BUILD_TYPE/UFO_BUILD_TYPE/CMAKE_BUILD_TYPE = release/release/Release
-- media -- LIBVA_INSTALL_PATH =
-- media -- MEDIA_VERSION = 22.6.0
-- media -- X11 Found
-- Checking for module 'libva-x11'
--   Found libva-x11, version 1.16.0
-- /home/intel/Desktop/dev_media/gfx-driver/Source/media-driver/media_driver/agnostic/common/vp/kernel/media_srcs.cmake doesn't exist, macro(media_include_subdirectory) just does nothing
-- /home/intel/Desktop/dev_media/gfx-driver/Source/media-driver/media_softlet/agnostic/common/media_interfaces/mmd/media_srcs.cmake doesn't exist, macro(media_include_subdirectory) just does nothing
-- LIBVA_DRIVERS_PATH = /usr/lib/x86_64-linux-gnu/dri
-- Configuring done
-- Generating done
-- Build files have been written to: /home/intel/Desktop/dev_media/gfx-driver/Source/build_media
root@intel-OptiPlex-3050:/home/intel/Desktop/dev_media/gfx-driver/Source/build_media#
root@intel-OptiPlex-3050:/home/intel/Desktop/dev_media/gfx-driver/Source/build_media# make -j8
Scanning dependencies of target iHD_drv_video_SSE2
Scanning dependencies of target iHD_drv_video_CP
Scanning dependencies of target iHD_drv_video_mos
Scanning dependencies of target igfxcmrt
Scanning dependencies of target iHD_drv_video_VP
Scanning dependencies of target iHD_drv_video_SSE4
Scanning dependencies of target iHD_drv_video_COMMON
Scanning dependencies of target iHD_drv_video_CODEC
[  0%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_printf_host.cpp.o
[  0%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_SSE2.dir/agnostic/common/cm/cm_mem_sse2_impl.cpp.o
[  0%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_SSE4.dir/linux/common/cm/hal/osservice/cm_mem_os_sse4_impl.cpp.o
[  0%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CP.dir/__/media_softlet/agnostic/common/cp/encodecp.cpp.o
[  0%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_rt_helpers.cpp.o
...
...
...
[ 27%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_VP.dir/__/media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp.o
[ 27%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_VP.dir/__/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline.cpp.o
...
...
...
[100%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/__/media_softlet/agnostic/common/codec/hal/enc/shared/mmc/encode_mem_compression.cpp.o
[100%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/__/media_softlet/agnostic/common/codec/hal/shared/codec_hw_next.cpp.o
[100%] Built target iHD_drv_video_CODEC
Scanning dependencies of target iHD_drv_video_STATIC
Scanning dependencies of target iHD_drv_video
[100%] Linking CXX shared library iHD_drv_video.so
[100%] Linking CXX static library iHD_drv_video.a
[100%] Built target iHD_drv_video
[100%] Built target iHD_drv_video_STATIC
~/build_media#

heitbaum, 2022-10-24:
Hi @cqian2 

here is the full log

```
BUILD      media-driver (target)
    TOOLCHAIN      cmake (auto-detect)
Executing (target): cmake -GNinja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/etc/cmake-x86_64-libreelec-linux-gnu.conf -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -DBUILD_CMRTLIB=OFF -DBUILD_KERNELS=ON -DBUILD_TYPE=release -DENABLE_NONFREE_KERNELS=ON -DMEDIA_BUILD_FATAL_WARNINGS=ON -DMEDIA_RUN_TEST_SUITE=OFF /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_INSTALL_PREFIX = /usr
-- Found PkgConfig: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/pkg-config (found version "0.29.2") 
-- Could NOT find X11 (missing: X11_X11_INCLUDE_PATH X11_X11_LIB) 
-- Checking for module 'igdgmm>=12.0.0'
--   Found igdgmm, version 12.2.0
-- media -- PLATFORM = linux
-- media -- ARCH = 64
-- media -- CMAKE_CURRENT_LIST_DIR = /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/media_driver
-- media -- INCLUDED_LIBS = 
-- media -- LIB_NAME = iHD_drv_video
-- media -- OUTPUT_NAME = 
-- media -- BUILD_TYPE/UFO_BUILD_TYPE/CMAKE_BUILD_TYPE = release/release/Release
-- media -- LIBVA_INSTALL_PATH = 
-- media -- MEDIA_VERSION = 22.6.0
-- /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/media_driver/agnostic/common/vp/kernel/media_srcs.cmake doesn't exist, macro(media_include_subdirectory) just does nothing
-- /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/media_softlet/agnostic/common/media_interfaces/mmd/media_srcs.cmake doesn't exist, macro(media_include_subdirectory) just does nothing
-- Checking for module 'libva>=1.8.0'
--   Found libva, version 1.16.0
-- LIBVA_DRIVERS_PATH = /usr/lib/dri
-- Configuring done
-- Generating done
-- Build files have been written to: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/.x86_64-libreelec-linux-gnu
Executing (target): ninja 
```
error
```
[1075/1159] Building CXX object media_driver/CMakeFiles/iHD_drv_...softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp.o
FAILED: media_driver/CMakeFiles/iHD_drv_video_VP.dir/__/media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp.o 
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-g++ -DCLASS_TRACE=0 -DENABLE_KERNELS -DHEVC_FEI_ENABLE_CMRT -DIGFX_DG2_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLN_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_DG1_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DIGFX_XEHP_SDV_CMFCPATCH_SUPPORTED -DIGFX_XEHP_SDV_SUPPORTED -DIGFX_XE_HPG_CMFCPATCH_SUPPORTED -DIGFX_XE_HPG_SUPPORTED -DMEDIA_VERSION=\"22.6.0\" -DMEDIA_VERSION_DETAILS=\"339d6f9c5a\" -DVEBOX_AUTO_DENOISE_SUPPORTED=1 -D_AV1_DECODE_SUPPORTED -D_AV1_ENCODE_VDENC_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_AVC_ENCODE_VME_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_DECODE_PROCESSING_SUPPORTED -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_HEVC_ENCODE_VME_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_MPEG2_ENCODE_VME_SUPPORTED -D_RELEASE -D_VC1_DECODE_SUPPORTED -D_VP8_DECODE_SUPPORTED -D_VP8_ENCODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/media_softlet/linux/common/os/i915/include -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/media_softlet/linux/common/os/i915/include/uapi -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/media_driver/agnostic/common/os -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.0/media_driver/linux/common/os [... cut ...]
 3433 |     curbe = (void*)(&HDRStatic);
      |     ~~~~~~^~~~~~~~~~~~~~~~~~~~~
../media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp:2817:34: note: 'HDRStatic' declared here
 2817 |     MEDIA_WALKER_HDR_STATIC_DATA HDRStatic           = g_cInit_MEDIA_STATIC_HDR;
      |                                  ^~~~~~~~~
../media_softlet/agnostic/common/vp/hal/packet/vp_render_hdr_kernel.cpp:2817:34: note: 'curbe' declared here
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

cqian2, 2022-10-25:
It seems that we only need to add "static" when declaring "HDRStatic", to fix this issue.
Add LhGu to help check whether or not this change will impact.

PR #1525 (2022-10-26): Fix hdr render issue
Fixed #1519
commit d879c147 (2022-10-26): Fix hdr render issue

Fixed #1519
closed 2022-10-26 by intel-mediadev