intel/media-driver #1521: [Bug]: What's the license status of media_driver/agnostic/Xe_M/Xe_XPM/vp/kernel/cmfcpatch/igvpkrn_xe_xpm_cmfcpatch.c?

sebastinas, 2022-10-23:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Same as #1517, but this time for `media_driver/agnostic/Xe_M/Xe_XPM/vp/kernel/cmfcpatch/igvpkrn_xe_xpm_cmfcpatch.c`.

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

No.

FurongZhang, 2022-10-23:
@LhGu , is it possible for you to follow up this issue?

Xiaogangli-intel, 2022-10-24:
Hi @sebastinas, could you teach us how you found these issues? We will add to our build in the same way, so we can catch issues earlier.

sebastinas, 2022-10-24:
The list also available from https://salsa.debian.org/multimedia-team/intel-media-driver/-/blob/master/debian/copyright

LhGu, 2022-10-24:
I confirmed all files in the list and maked a PR to fix it.

sebastinas, 2022-10-24:
22.6.0 together with this patch makes it work for the Debian package. Thanks @LhGu.

LhGu, 2022-10-25:
> 22.6.0 together with this patch makes it work for the Debian package. Thanks @LhGu.

Thanks for your double confirm, and we will merge it ASAP.

PR #1523 (2022-10-25): Fix build failure with full free kernel
Fixed #1521
commit e5bb9039 (2022-10-25): Fix build failure with full free kernel

Fixed #1521
closed 2022-10-25 by intel-mediadev