intel/media-driver #1522: [Feature]: Compile-time enable/disable codecs support

frantisekz, 2022-10-24:
### What Feature?

I'd like to be able to disable support for various codecs compile-time. This would allow the media-driver to be included in Fedora to provide vaapi acceleration at least for free and not patent-encumbered formats like VP8, VP9 and AV1. Users would still be able to choose full-featured version from third-party repository.

### What's the usage scenario would be benifited?

Others

### What impacted?

I guess basically everything from the list.

eero-t, 2023-01-18:
@XinfengZhang / @Jexu Any progress on this?

frantisekz, 2023-01-18:
I was able to achieve this with:

  -DVC1_Decode_Supported="no" \
  -DAVC_Encode_VME_Supported="no" \
  -DAVC_Encode_VDEnc_Supported="no" \
  -DHEVC_Encode_VME_Supported="no" \
  -DHEVC_Encode_VDEnc_Supported="no" \
  -DAVC_Decode_Supported="no" \
  -DHEVC_Decode_Supported="no" \
  -DAVC_Decode_Supported="no" \
  -DMPEG2_Decode_Supported="no" \

Which results with the driver exposing just VP8, VP9, AV1, and JPEG. It, however, misses encoding completely when built this way.

XinfengZhang, 2023-05-08:
#1643

XinfengZhang, 2023-05-08:
@stellawuintel , could you help to paste the commit and close this issue?

closed 2023-05-08 by intel-mediadev