intel/media-driver #1524: [Bug]: build error with libva disable X11

wenqingx, 2022-10-25:
### Which component impacted?

Build

### What happened?

libva: https://github.com/intel/libva/commit/da4a18fe814c9695f420d466ad3065cebdbe288a 
gmmlib: https://github.com/intel/gmmlib/commit/416ee5042f9c192344650cf90d16594007dc72e1 
dirver:https://github.com/intel/media-driver/commit/4a5afc5e15d67f70ba0e2cb7916b2cde8502a4a6

If libva does not enable X11 when compiling the driver, you will encounter a build error.
The configuration I just pulled is the latest one, which I reproduced on TGL.
But it seems that this has nothing to do with the platform

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

media-driver can build without X11.

Jexu, 2022-10-25:
It is same issue as https://github.com/intel/media-driver/issues/1472?

wenqingx, 2022-10-25:
> It is same issue as #1472?

It looks similar,If I add "-- enable - x11" when config and build libva, this issue will be fixed.

Jexu, 2022-10-25:
X11 found on local machine, but libva and media driver have different build option that previous one disables X11 while media driver enables it by default.

wenqingx, 2022-10-26:
> X11 found on local machine, but libva and media driver have different build option that previous one disables X11 while media driver enables it by default.

You mean that the driver will not force the X11 environment during compilation. Because X11 is enable on by default, the issue will fix if I disable X11 during compilation? thanks

nyanmisaka, 2022-10-26:
You can apply this patch and try again.
https://github.com/intel/media-driver/pull/1527.patch

wenqingx, 2022-10-26:
> You can apply this patch and try again. https://github.com/intel/media-driver/pull/1527.patch

OK, I will try to apply the patch ,thanks.

wenqingx, 2022-10-26:
> 
apply your patch , the media-driver build fixed with libva not enable x11. thanks your fixed patch.

wenqingx, 2022-10-26:
> You can apply this patch and try again. https://github.com/intel/media-driver/pull/1527.patch

apply your patch , the media-driver build fixed with libva not enable x11. thanks your fixed patch.

PR #1527 (2022-10-31): Fix build that x11 exists but libva-x11 is disabled
```
-- media -- X11 Found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
```

Fixes #1524 #1472