intel/media-driver #1530: Difficulties to Install on ubuntu 18.04

MyraBaba, 2022-10-27:
### Which component impacted?

Decode, Encode, Video Processing

### What happened?

I compiled the lates version todat both required libva and gmmlib . thann compile and install medai-driver succesfully.

when I wrote 👍 

>  bc@bcl:~/toolscompile/build_media$ export LIBVA_DRIVERS_PATH=/usr/local/lib/dri/iHD_drv_video.so
bc@bcl:~/toolscompile/build_media$ export LIBVA_DRIVER_NAME=iHD
bc@bcl:~/toolscompile/build_media$ vainfo
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

cpu :  Intel(R) Xeon(R) E-2276M

I am unsuccessful to use HW acceleration for decoding . Almost giving up.

we are trying to decode 4 x 1080p 15fps camera rtsp stream with HW accelerated .

Best 

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery, Playback

### Debug Information

libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so/iHD_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

dvrogozh, 2022-10-27:
Can you, please, check where driver got installed? Provide output of:
```
ls /usr/local/lib/dri/iHD_drv_video.so
find /use -name iHD_drv_video.so
```

This path in the "debug information output" does not make sense:
```
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so/iHD_drv_video.so
```

To understand your setup, please, provide how projects were configured, i.e. full ./autogen.sh or cmake or meson configuration lines.

MyraBaba, 2022-10-27:
@dvrogozh 

> 
> ls -l /usr/local/lib/dri/iHD_drv_video.so
> -rwxr-xr-x 1 root root 326235880 Eki 27 13:49 /usr/local/lib/dri/iHD_drv_video.so

HW : Lenovo p73 Laptop ubuntu 18.04 
cpu : Intel(R) Xeon(R) E-2276M  CPU @ 2.80GHz

we want to change our jetson base system to intel if we succeed to decode 4 X rtsp h264 streams  on the 4 core atom or celeron systems without much usage of the cpu .

Is there any benchmark for CPU's to decoding rtsp h264 decoding for using with opencv

MyraBaba, 2022-10-27:
> dmesg | grep drm
[    2.214205] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    2.214205] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 0

bc@bcl:~$ uname -a
Linux bcl 4.18.0-15-generic #16~18.04.1-Ubuntu SMP Thu Feb 7 14:06:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
bc@bcl:~$ sudo cat /proc/cpuinfo | grep "model name" | uniq
model name	: Intel(R) Xeon(R) E-2276M  CPU @ 2.80GHz

bc@bcl:~$ lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:3ec4] (rev 0d)
00:01.0 PCI bridge [0604]: Intel Corporation Skylake PCIe Controller (x16) [8086:1901] (rev 0d)
00:04.0 Signal processing controller [1180]: Intel Corporation Skylake Processor Thermal Subsystem [8086:1903] (rev 0d)
00:08.0 System peripheral [0880]: Intel Corporation Skylake Gaussian Mixture Model [8086:1911]
00:12.0 Signal processing controller [1180]: Intel Corporation Device [8086:a379] (rev 10)
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:a36d] (rev 10)
00:14.2 RAM memory [0500]: Intel Corporation Device [8086:a36f] (rev 10)
00:15.0 Serial bus controller [0c80]: Intel Corporation Device [8086:a368] (rev 10)
00:15.1 Serial bus controller [0c80]: Intel Corporation Device [8086:a369] (rev 10)
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:a360] (rev 10)
00:16.3 Serial controller [0700]: Intel Corporation Device [8086:a363] (rev 10)
00:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:a340] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:a338] (rev f0)
00:1c.5 PCI bridge [0604]: Intel Corporation Device [8086:a33d] (rev f0)
00:1c.7 PCI bridge [0604]: Intel Corporation Device [8086:a33f] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:a330] (rev f0)
00:1e.0 Communication controller [0780]: Intel Corporation Device [8086:a328] (rev 10)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a30e] (rev 10)
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:a348] (rev 10)
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:a323] (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device [8086:a324] (rev 10)
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (7) I219-LM [8086:15bb] (rev 10)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1eb5] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10f8] (rev a1)
01:00.2 USB controller [0c03]: NVIDIA Corporation Device [10de:1ad8] (rev a1)
01:00.3 Serial bus controller [0c80]: NVIDIA Corporation Device [10de:1ad9] (rev a1)
02:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd Device [144d:a808]
04:00.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
05:00.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
05:01.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
05:02.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
05:04.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
06:00.0 System peripheral [0880]: Intel Corporation JHL7540 Thunderbolt 3 NHI [Titan Ridge 4C 2018] [8086:15eb] (rev 06)
2c:00.0 USB controller [0c03]: Intel Corporation JHL7540 Thunderbolt 3 USB Controller [Titan Ridge 4C 2018] [8086:15ec] (rev 06)
52:00.0 Network controller [0280]: Intel Corporation Device [8086:2723] (rev 1a)
54:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader [10ec:525a] (rev 01)
55:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd Device [144d:a808]

bc@bcl:~$ cat /sys/class/drm/card0/device/vendor
0x10de

MyraBaba, 2022-10-27:
@dvrogozh  meanwhile is it worth it to use QuickVideo intel HW cpu acceleration ? is there any cpu usage or how much less than the sw decoding?

MyraBaba, 2022-10-27:
@dvrogozh  
 We will possibly use below CPU's

Where we can find the h264 rtsp decoding performance benchmark ? At the the media-driver or  Vaap onevpl will work. 

1080p and 15 fps each frame 4 camera.

And also we need min %50 of the cpu for other jobs.

Intel Atom® x7-E3950 Processor
 Intel Atom Elkhart Lake CPU, 1.3-2.0 GHz  
Intel Atom Apollo Lake CPU, 1.3-1.5 GHz

dvrogozh, 2022-10-27:
Ok, so, your system:
1. Has Xeon CPU with Intel graphics (Coffee Lake): https://ark.intel.com/content/www/us/en/ark/products/192989/intel-xeon-e2276m-processor-12m-cache-2-80-ghz.html
2. Has NVidia card installed.

Intel GPU was **not** detected on your system. It is missing in lspci list (grep for VGA, it shows only nvidia) and eventually is missed in dmesg kernel log (for Intel we would see lines with 'i915' driver which is Intel kernel mode driver).

Suggestions why Intel GPU is not seen:
1. Check BIOS settings. It might be that Intel GPU is simply switched off in BIOS. If so, switch it on and try again to check `lspic -nn`  for Intel VGA line and `dmesg | grep drm` for i915.
2. Since you have nvidia, did you install any nvidia drivers? I wonder whether they conflict with Intel GPU support. It might worth to try on clean ubuntu installation w/o installing nvidia drivers. But first, I would definitely check BIOS.

So, before trying out vainfo, i.e. user space media stack, you need to make sure that Intel GPU is there and support on kernel mode side.

dvrogozh, 2022-10-27:
> Is there any benchmark for CPU's to decoding rtsp h264 decoding for using with opencv

The benchmarks I can point to in open source concerns full transcoding pipeline for certain GPUs (which are: Gen9 series (CoffeeLake, KabyLake, SkyLake), DG1 and ATS-M) with a focus on transcoding performance (for DG1 and ATS-M) and encoding quality comparing to x264/x265 software encodings (for Gen9, DG1, ATS-M). These benchmarks are available here: https://github.com/intel/media-delivery/tree/master/doc/benchmarks. Gen9 is hidden under "VCA" on https://github.com/intel/media-delivery/blob/master/doc/benchmarks/intel-iris-xe-max-graphics/intel-iris-xe-max-graphics.md.

HW acceleration should bring down CPU% comparing to software codecs and have better performance.

XinfengZhang, 2022-12-06:
@MyraBaba , looks the issue is that there are no intel gpu found in OS layer. 
just like the comments: need to check the bios setting to enable integrated gpu , and also need co check whether i915 is enabled. 
if no other concern , could we close the issue?

closed 2023-05-06 by XinfengZhang