intel/media-driver #1531: Question on debugging vaapi for a new Chromium patch

gustavqmatic, 2022-10-28:
I’m trying to figure out why the pixel format from Chromium used in this patch (https://chromium-review.googlesource.com/c/chromium/src/+/3646633 for issue https://bugs.chromium.org/p/chromium/issues/detail?id=1326754) does not work with Intel media driver when I test it.

I can see that the request on vaapi is handled as a VAContext "decoder" and it fails in EndPicture where it fails because Chromium sets format RGBX_8888/XRGB while the driver only handles YUV_420_BIPLANAR/NV12.

One idea I want to debug/verify/try is if the context might be wrong, that is should be in VAContext "VP" instead of "decoder". In that case it would use a different EndPicture that does not seem to have this check for NV12. It would also explain why other seems to get it to work. Does that makes sense?

What is controlling the context? How can I change it? Or could something else be changed to allow vaapi calls to accept RGBX_8888/XRGB?

I build it using Yocto (kirkstone with meta-intel) in Wayland/weston.

gustavqmatic, 2022-11-02:
A bit of update from my side. I believe the issue is not with the context, it is decode and it makes perfect sense that is also what I see.

But my question is still if/how XRGB is supported in the Codec. I see for example that AVC seems to only support NV12, in fact the list of supported formats (https://github.com/intel/media-driver/blob/master/docs/media_features.md) does not mention XRGB anywhere.

Can anyone comment on RGBX_8888/XRGB support in media-driver?

XinfengZhang, 2022-11-03:
no RGBX support from decoding , encoding support it , you could use vaQuerySurfaceAttributes to get the format supported list for given codec & encode or decode.

XinfengZhang, 2022-11-17:
maybe you should try vpp context. may we close this issue?

gustavqmatic, 2022-11-17:
Yes, we can close the issue. I got an answer to my question.

But I find it strange that one Intel team (Chromium developer [jianh...@intel.com]) develop a patch for Chromium that only have support for RGBX_8888 while another (your team developing the Intel media driver) only have support for NV12. Do you have communication within Intel between these two teams?

Jexu, 2022-11-22:
> Yes, we can close the issue. I got an answer to my question.
> 
> But I find it strange that one Intel team (Chromium developer [[jianh...@intel.com](mailto:jianh...@intel.com)]) develop a patch for Chromium that only have support for RGBX_8888 while another (your team developing the Intel media driver) only have support for NV12. Do you have communication within Intel between these two teams?

If the development is based on media driver, we suggest querying surface attributes firstly to get supported format.  This maybe the case that unsupported format causing media not work.

I will close this issue now, you can re-open it again if having any other questions. Thanks.

closed 2022-11-22 by Jexu