intel/media-driver #1532: [Bug]: encode breaks with dynamic resolution changes

Ambyjkl, 2022-10-28:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

1. In Linux, install media-driver, intel-vaapi-driver, and gstreamer (versions listed in the debug info section)
2. Run the following gstreamer pipeline, where the input file `in.264` has varying resolution:
```
GST_DEBUG=vaapi*:2 gst-launch-1.0 filesrc location=in.264 ! h264parse ! avdec_h264 ! videoconvert ! vaapih264enc ! filesink location=out.264
```
3. The following errors are thrown, and the pipeline terminates right at the first resolution change, as seen by playing the output file
```
0:00:00.115816650 68995 0x55d2787e7240 ERROR                  vaapi gstvaapiencoder.c:348:gst_vaapi_encoder_encode_and_queue: failed to encode frame (status = -1)
0:00:00.115829647 68995 0x55d2787e7240 ERROR            vaapiencode gstvaapiencode.c:694:gst_vaapiencode_handle_frame: failed to encode frame 23 (status -1)
ERROR: from element /GstPipeline:pipeline0/GstH264Parse:h264parse0: Internal data stream error.
Additional debug info:
../gstreamer/subprojects/gstreamer/libs/gst/base/gstbaseparse.c(3681): gst_base_parse_loop (): /GstPipeline:pipeline0/GstH264Parse:h264parse0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.052042037
Setting pipeline to NULL ...
Freeing pipeline ...
```
4. However, upon forcing the use of the older intel-vaapi-driver (i965), it succeeds, and the output is as expected, making it kind of a regression from the older driver behaviour
```
LIBVA_DRIVER_NAME=i965 GST_DEBUG=vaapi*:2 gst-launch-1.0 filesrc location=in.264 ! h264parse ! avdec_h264 ! videoconvert ! vaapih264enc ! filesink location=out.264
```
5. Using VP8 or VP9 instead of H264 didn't work either on media-driver, but again did work on intel-vaapi-driver, confirming it's not a codec-related issue.
6. I also tried the latest git version of media-driver, which didn't work either, so it's not already fixed.

The tests were performed on an i7 8700K CPU with UHD Graphics 630, and also partly on Alder Lake i5 12400 (UHD Graphics 730) and 12500 (UHD Graphics 770), where I could not test intel-vaapi-driver because it doesn't support it.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

1. Package versions (I'm using Arch Linux)
```
intel-media-driver 22.4.4-2
intel-media-driver-git 2022.6.0.r71.g0e202e11d-1
libva-intel-driver 2.4.1-2
gstreamer 1.20.4-1
gst-plugins-base 1.20.4-1
gst-plugins-base-libs 1.20.4-1
gst-plugins-good 1.20.4-1
gst-plugins-bad 1.20.4-1
gst-plugins-bad-libs 1.20.4-1
gst-plugins-ugly 1.20.4-1
gst-libav 1.20.4-1
gstreamer-vaapi 1.20.4-1
```
2. Yes
3. vainfo, with media-driver (`iHD`): [vainfo_iHD.log](https://github.com/intel/media-driver/files/9892247/vainfo_iHD.log)
vainfo, with intel-vaapi-driver (`i965`): [vainfo_i965.log](https://github.com/intel/media-driver/files/9892249/vainfo_i965.log)
4. Logs generated when running `GST_DEBUG=vaapi*:2 gst-launch-1.0 filesrc location=in.264 ! h264parse ! avdec_h264 ! videoconvert ! vaapih264enc ! filesink location=out.264`:
[libva_trace.log.221747.thd-0x000118e6](https://github.com/intel/media-driver/files/9892260/libva_trace.log.221747.thd-0x000118e6.txt)
[libva_trace.log.221747.thd-0x000118e7](https://github.com/intel/media-driver/files/9892261/libva_trace.log.221747.thd-0x000118e7.txt)
[libva_trace.log.221747.thd-0x000118f5](https://github.com/intel/media-driver/files/9892262/libva_trace.log.221747.thd-0x000118f5.txt)
5. No GPU hang

Also here is the input file `in.264` I'm using (zipped up so GitHub lets me attach it): 
[in.264.zip](https://github.com/intel/media-driver/files/9892269/in.264.zip)

intel-mediadev, 2022-11-18:
Auto Created VSMGWL-59151 for further analysis.

Ambyjkl, 2023-01-03:
@Tianhaol what's the status on this issue? Is it possible to expedite it, it's now impacting production on our end.

dmenshov, 2023-06-05:
There should be the config reset in case dynamic resolution change. It is relible behaviour in case of reconfiguration. Since i965 has different pipeline for context creation the dynamic resolution change works.

Ambyjkl, 2023-06-05:
@dmenshov thanks, I'll see if that works

dmenshov, 2023-06-05:
Logs after change source code at gstreamer:
root:/home/gta/dmenshov# export LIBVA_DRIVER_NAME=iHD
root:/home/gta/dmenshov# gst-launch-1.0 filesrc location=in.264 ! h264parse ! avdec_h264 ! videoconvert ! vaapih264enc ! filesink location=out.264
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva info: va_openDriver() returns 0
Setting pipeline to PAUSED ...
MobaXterm X11 proxy: Unsupported authorisation protocol
MobaXterm X11 proxy: Unsupported authorisation protocol
Pipeline is PREROLLING ...
Got context from element 'vaapiencodeh264-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm1", gst.vaapi.Display.GObject=(GstObject)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm1";
Redistribute latency...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
Redistribute latency...
Redistribute latency...
Redistribute latency...
Redistribute latency...
Redistribute latency...14.7 %)
Redistribute latency...
Redistribute latency...20.0 %)
Redistribute latency...57.9 %)
Redistribute latency...
Redistribute latency...73.6 %)
Got EOS from element "pipeline0".
Execution ended after 0:00:00.679988246
Setting pipeline to NULL ...
Freeing pipeline ...
root:/home/gta/dmenshov

dmenshov, 2023-06-15:
What should we do with the task? Is it actuall?

Ambyjkl, 2023-06-15:
Yeah, I can confirm that the solution works, thanks @dmenshov, I'll close this issue.

closed 2023-06-15 by Ambyjkl