intel/media-driver #1533: [Bug]:

quentinschibler, 2022-10-30:
### Which component impacted?

Build

### What happened?

Using musl as libc, I am not able to build because execinfo.h is not found.

### What's the usage scenario when you are seeing the problem?

Others

### Debug Information

Here is the output of the build :

media_softlet/linux/common/os/osservice/mos_utilities_specific.cpp:37:10: fatal error: execinfo.h: No such file or directory

quentinschibler, 2022-10-30:
Here is the modified patch, working on master :
[execinfo.txt](https://github.com/intel/media-driver/files/9896301/execinfo.txt)

bkuhls, 2022-10-31:
duplicate of https://github.com/intel/media-driver/pull/1437, had no time to look at FindBacktrace yet

dvrogozh, 2022-10-31:
> Using musl as libc,

@quentinschibler : what is musl? Please, be specific and provide:
1. OS you are working on
2. Additional packages you've installed
3. cmake configuration command line

I guess there is specific way you've installed "musl" and added cmake configuration - that's essential for the bug description.

quentinschibler, 2022-10-31:
> duplicate of #1437, had no time to look at FindBacktrace yet

Sorry, I searched for musl in the issue tracker, and had not found anything. Probably because the issue was using uClibc. You can close the issue then.

quentinschibler, 2022-10-31:
> @quentinschibler : what is musl? Please, be specific and provide:
>
>     1. OS you are working on
>
>     2. Additional packages you've installed
>
>     3. cmake configuration command line
>
>
> I guess there [...]

os is linux.
musl is a libc implementation, the installation process is somewhat similar to glibc, and is not relevant to the issue. The issue is that execinfo, which a glibc specific header that is NOT part of the libc specification, is used without checking for glibc in CMakeList.txt.

dvrogozh, 2022-10-31:
> Musl is a libc implementation, the installation process is somewhat similar to glibc, and is not relevant to the issue.

It is relevant. You file an issue to open source project. How issue can be reproduced? That's needed at least to make sure that the provided fix won't be blind and can be verified.

quentinschibler, 2022-10-31:
If you want to reproduce the issue, you'd be better off installing a linux distro using musl. If you absolutely want to install musl on an existing system, then you can follow the "building a cross compiler targeting musl libc" [here](https://wiki.musl-libc.org/getting-started.html).

dvrogozh, 2022-10-31:
This will do. Thank you.

PR #1536 (2022-11-16): build: check for backtrace(3) via FindBacktrace
Fixes: #1533

Change-Id: I81bcdb5e65c26f94b174af9b5e6218c14601c50f
commit be887135 (2022-11-16): build: check for backtrace(3) via FindBacktrace

Fixes: #1533

Change-Id: I81bcdb5e65c26f94b174af9b5e6218c14601c50f
closed 2022-11-16 by intel-mediadev