intel/media-driver #1534: [Feature]: OS Guidance

colemickens, 2022-10-30:
### What Feature?

Hi,

I'm trying to cleanup GPU options for NixOS. How can we best support media-driver?

How does mesa behave when both intel-vaapi-driver and intel-media-driver are available? Is there a way to set a preference without explicitly overriding the LIBVA driver name in general, maybe so as-to still allow for any fallback behavior for multi-GPU scenarios?

### What's the usage scenario would be benifited?

Others

colemickens, 2022-10-30:
alright, I'm guessing that it tries and prefers iHD if available, but confirmation would be great: https://github.com/intel/libva/commit/09c8c2e20b7763fc44299172dbb16061b57f9c71#diff-4f0deceafa6df3a50bc6e10cf27e036f0828b7cc0ff254c720f5115ecb261d04R39

nyanmisaka, 2022-11-02:
- Mesa only provide VAAPI driver for AMD, not Intel.
- On Intel, `vainfo` will try iHD first, if it fails then try i965.
- AFAIK there no implicit fallback in VAAPI unless handle it manually.
Here's an example in [ffmpeg](https://github.com/FFmpeg/FFmpeg/blob/12733c0cbd49077d3aac48007f674f14d1e15ccd/libavutil/hwcontext_vaapi.c#L1651-L1804).

XinfengZhang, 2022-11-16:
you will open the device ,get fd, for each fd it will try ihd then i965(if it is Intel device)

XinfengZhang, 2022-11-16:
suppose your questions is addressed,any other problem,feel free to reopen it

closed 2022-11-16 by XinfengZhang