intel/media-driver #1537: [Bug]: vaGetImage is very slow on dGPU for 4K 10bit clip

mengker33, 2022-11-03:
### Which component impacted?

Decode

### What happened?

1. On linux, using gstreamer va plugins to decode a 4K p010 format bitstream:<pre> gst-launch-1.0 -vf filesrc location=4k_p010.h265 ! h265parse ! vah265dec ! fakesink</pre>
2. The whole process for running the cmd line above is extremely slow (**~3sec for one frame**). Debugging into it, I found the slow step happened at GPU to CPU copy using vaGetImage. I don't see it slow on iGPU. Also, when I tried to use vaDeriveImage, the speed is normally quick.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Decoding a 4K ( with formats expect NV12) clip is a quite basic function. We expect faster decoding on dGPU instead of much slower than iGPU.

Jexu, 2022-11-03:
Could you help check if blt copy in 'SwizzleSurface' or cpu copy in 'DdiMedia_CopyPlane' cost the time?

mengker33, 2022-11-03:
> Could you help check if blt copy in 'SwizzleSurface' or cpu copy in 'DdiMedia_CopyPlane' cost the time?

I think yes. I have seen not-NV12 format will go into SwizzleSurface, then there is a call for **cpublt** is quite slow.

intel-mediadev, 2022-11-03:
Auto Created VSMGWL-58535 for further analysis.

uartie, 2022-11-07:
FFmpeg vaapi 10bit decode is very slow, too, on dGPU.

Jexu, 2022-11-08:
@cqian2  could you help take a look this issue about blt copy.

hye5, 2022-11-17:
What platform and is it a regression?

XinfengZhang, 2022-11-17:
obviously, it is ACM

MicroYY, 2022-12-04:
Hi @mengker33 Pls try https://github.com/intel/media-driver/pull/1569 to see if the issue still exists.

mengker33, 2022-12-05:
> Hi @mengker33 Pls try #1569 to see if the issue still exists.

The issue is still there after this patch applied. Removing "image->format.fourcc != VA_FOURCC_NV12" may not help from my understanding, because p010 format always goto "if" path no matter "image->format.fourcc != VA_FOURCC_NV12" removed or not.

MicroYY, 2022-12-05:
@mengker33 Refined the PR. Could you pls retry?

mengker33, 2022-12-05:
> @mengker33 Refined the PR. Could you pls retry?

It is still quite slow.

PR #1569 (2022-12-16): [Media Common] vaGetImage optimization on D-card.
Fix #1537

On discrete cards which have local memory, map shadow buffer instead of surface bo. Otherwise cpu memcpy will be very slow.
commit b67820f3 (2022-12-16): [Media Common] vaGetImage optimization on D-card.

Fix #1537

On discrete cards which have local memory, map shadow buffer instead of
surface bo. Otherwise cpu memcpy will be very slow.
closed 2022-12-16 by intel-mediadev