intel/media-driver #1538: [Feature]: VPP 12-bit YUV Support

TheanSiew, 2022-11-03:
### What Feature?

Both ffmpeg-vaapi and gst-va/gst-vaapi are using 12bit fourcc codes instead of 16bit fourcc codes. Transcoding pipeline will be broken if no 12-bit support in vpp. Request to have the following support combination in media-driver:

rt_format=VA_RT_FORMAT_YUV420_12 and fourcc=p012 for YUV 4:2:0 12-bit,
rt_format=VA_RT_FORMAT_YUV422_12 and fourcc=y212  for YUV 4:2:2 12-bit
rt_format=VA_RT_FORMAT_YUV444_12 and fourcc=y412  for YUV 4:4:4 12-bit.

### What's the usage scenario would be benifited?

Transcode for media delivery

### What impacted?

 ffmpeg-vaapi and gst-va/gst-vaapi transcoding pipeline is currently broken. Request to have the support implemented in Q4'22.

FurongZhang, 2022-11-08:
Estimated. WIP on implementation.

FurongZhang, 2022-11-21:
https://github.com/intel/media-driver/pull/1552/files

FurongZhang, 2022-12-12:
We have implemented this feature and validated via GStreamer. 12bit VPP will be supported in 2022 Q4 media driver release. Hence, I close this and please feel free to let me know if any questions, thank you!

closed 2022-12-12 by FurongZhang