intel/media-driver #1540: [Bug]: Building media-driver with {AVC,HEVC}Encode_{VME,VDEnc}_Supported="no" results in linking failure

frantisekz, 2022-11-03:
### Which component impacted?

Build

### What happened?

Build of media-driver 22.6.1 with:

  -DENABLE_NONFREE_KERNELS:BOOL=OFF \
  -DENABLE_KERNELS:BOOL=OFF \
  -DBUILD_KERNELS:BOOL=OFF \
  -DMEDIA_BUILD_FATAL_WARNINGS=OFF \
  -DBUILD_SHARED_LIBS:BOOL=OFF \
  -DGEN10:BOOL=OFF \
  -DAVC_Encode_VME_Supported="no" \
  -DAVC_Encode_VDEnc_Supported="no" \
  -DHEVC_Encode_VME_Supported="no" \
  -DHEVC_Encode_VDEnc_Supported="no" \
  -G Ninja

results with linker failure:

```
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:568: warning: type 'struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS' violates the C++ One Definition Rule [-Wodr]
  568 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_softlet/agnostic/common/vp/hal/packet/vp_vebox_common.h:31: note: a different type is defined in another translation unit
   31 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:570: note: the first difference of corresponding definitions is field 'pSurfInput'
  570 |     PVPHAL_SURFACE                  pSurfInput;
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_softlet/agnostic/common/vp/hal/packet/vp_vebox_common.h:33: note: a field of same name but different type is defined in another translation unit
   33 |     PVP_SURFACE                     pSurfInput;
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:568: note: type 'struct VPHAL_SURFACE *' should match type 'struct VP_SURFACE *'
  568 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/usr/bin/ld: /tmp/ccSE7Y2w.ltrans35.ltrans.o: in function `CodechalInterfacesXe_Xpm::Initialize(void*, void*, MhwInterfaces*, _MOS_INTERFACE*)':
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_softlet/agnostic/common/os/mos_utilities.h:2834: undefined reference to `CodechalHwInterfaceXe_Xpm::CodechalHwInterfaceXe_Xpm(_MOS_INTERFACE*, _CODECHAL_FUNCTION, MhwInterfaces*, bool)'
/usr/bin/ld: /tmp/ccSE7Y2w.ltrans36.ltrans.o: in function `CodechalInterfacesXe_Hpm::CreateCodecHalInterface(MhwInterfaces*, CodechalHwInterface*&, CodechalDebugInterface*&, _MOS_INTERFACE*, _CODECHAL_FUNCTION, bool)':
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_softlet/agnostic/common/os/mos_utilities.h:2834: undefined reference to `CodechalHwInterfaceXe_Hpm::CodechalHwInterfaceXe_Hpm(_MOS_INTERFACE*, _CODECHAL_FUNCTION, MhwInterfaces*, bool)'
/usr/bin/ld: /tmp/ccSE7Y2w.ltrans53.ltrans.o: in function `CodechalEncoderState::Allocate(CodechalSetting*)':
/builddir/build/BUILD/media-driver-intel-media-22.6.1/media_softlet/agnostic/common/os/mos_utilities.h:2834: undefined reference to `CodechalEncodeTrackedBufferHevc::CodechalEncodeTrackedBufferHevc(CodechalEncoderState*)'
/usr/bin/ld: /tmp/ccSE7Y2w.ltrans56.ltrans.o:(.data.rel.ro+0x40): undefined reference to `CodechalCmdInitializer::CmdInitializerSetDmem(bool)'
/usr/bin/ld: /tmp/ccSE7Y2w.ltrans58.ltrans.o:(.data.rel.ro+0x48): undefined reference to `CodechalCmdInitializerG11::CmdInitializerSetDmem(bool)'
/usr/bin/ld: /tmp/ccSE7Y2w.ltrans58.ltrans.o:(.data.rel.ro+0x78): undefined reference to `CodechalCmdInitializerG11::AddCmdConstData(unsigned short, unsigned int*, unsigned short, unsigned int)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Entire build log:
[build.zip](https://github.com/intel/media-driver/files/9933775/build.zip)

### What's the usage scenario when you are seeing the problem?

Others

frantisekz, 2022-11-10:
The 22.6.2 tag fails in the similar way with some more errors:

```
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:568: warning: type 'struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS' violates the C++ One Definition Rule [-Wodr]
  568 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_softlet/agnostic/common/vp/hal/packet/vp_vebox_common.h:31: note: a different type is defined in another translation unit
   31 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:570: note: the first difference of corresponding definitions is field 'pSurfInput'
  570 |     PVPHAL_SURFACE                  pSurfInput;
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_softlet/agnostic/common/vp/hal/packet/vp_vebox_common.h:33: note: a field of same name but different type is defined in another translation unit
   33 |     PVP_SURFACE                     pSurfInput;
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:568: note: type 'struct VPHAL_SURFACE *' should match type 'struct VP_SURFACE *'
  568 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/usr/bin/ld: /tmp/ccu0lSzs.ltrans34.ltrans.o: in function `CodechalInterfacesXe_Xpm::Initialize(void*, void*, MhwInterfaces*, _MOS_INTERFACE*)':
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_softlet/agnostic/common/os/mos_utilities.h:2846: undefined reference to `CodechalHwInterfaceXe_Xpm::CodechalHwInterfaceXe_Xpm(_MOS_INTERFACE*, _CODECHAL_FUNCTION, MhwInterfaces*, bool)'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans35.ltrans.o: in function `CodechalInterfacesXe_Hpm::CreateCodecHalInterface(MhwInterfaces*, CodechalHwInterface*&, CodechalDebugInterface*&, _MOS_INTERFACE*, _CODECHAL_FUNCTION, bool)':
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_softlet/agnostic/common/os/mos_utilities.h:2846: undefined reference to `CodechalHwInterfaceXe_Hpm::CodechalHwInterfaceXe_Hpm(_MOS_INTERFACE*, _CODECHAL_FUNCTION, MhwInterfaces*, bool)'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o: in function `CodechalInterfacesXe_Lpm_Plus::Initialize(void*, void*, MhwInterfacesNext*, _MOS_INTERFACE*)':
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus/codec/hal/dec/jpeg/decode_jpeg_pipeline_adapter_xe_lpm_plus.cpp:32: undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::DecodeJpegPipelineAdapterXe_Lpm_Plus_Base(CodechalHwInterfaceNext*, CodechalDebugInterface*)'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x70): undefined reference to `typeinfo for DecodeJpegPipelineAdapterXe_Lpm_Plus_Base'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x420): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::Allocate(CodechalSetting*)'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x428): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::BeginFrame()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x430): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::EndFrame()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x438): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::Execute(void*)'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x440): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetStatusReport(void*, unsigned short)'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x448): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::Destroy()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x468): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::IsIncompletePicture()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x470): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::IsIncompleteJpegScan()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x478): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDummyReference()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x480): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDummyReferenceStatus()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x488): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::SetDummyReferenceStatus(_CODECHAL_DUMMY_REFERENCE_STATUS)'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x490): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetCompletedReport()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x498): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDecodeContext()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans52.ltrans.o:(.data.rel.ro+0x4a0): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDecodeContextHandle()'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans58.ltrans.o: in function `CodechalEncoderState::Allocate(CodechalSetting*)':
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_softlet/agnostic/common/os/mos_utilities.h:2846: undefined reference to `CodechalEncodeTrackedBufferHevc::CodechalEncodeTrackedBufferHevc(CodechalEncoderState*)'
[... cut ...] PipelineAdapterXe_Lpm_Plus_Base'
/usr/bin/ld: /tmp/ccu0lSzs.ltrans86.ltrans.o: in function `DecodeJpegPipelineAdapterXe_Lpm_Plus::~DecodeJpegPipelineAdapterXe_Lpm_Plus() [clone .lto_priv.0]':
/builddir/build/BUILD/media-driver-intel-media-22.6.2/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/codec/hal/dec/jpeg/pipeline/decode_jpeg_pipeline_adapter_xe_lpm_plus_base.h:41: undefined reference to `vtable for DecodeJpegPipelineAdapterXe_Lpm_Plus_Base'
collect2: error: ld returned 1 exit status
```

intel-mediadev, 2022-11-16:
Auto Created VSMGWL-59041 for further analysis.

frantisekz, 2023-01-13:
@bai-isaac Thanks for the PR!, However, I've tried to build 22.6.6 with #1594  applied on top of it and the result was similar failure:

```
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:568: warning: type 'struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS' violates the C++ One Definition Rule [-Wodr]
  568 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_softlet/agnostic/common/vp/hal/packet/vp_vebox_common.h:31: note: a different type is defined in another translation unit
   31 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:570: note: the first difference of corresponding definitions is field 'pSurfInput'
  570 |     PVPHAL_SURFACE                  pSurfInput;
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_softlet/agnostic/common/vp/hal/packet/vp_vebox_common.h:33: note: a field of same name but different type is defined in another translation unit
   33 |     PVP_SURFACE                     pSurfInput;
      | 
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_driver/agnostic/common/vp/hal/vphal_render_vebox_base.h:568: note: type 'struct VPHAL_SURFACE *' should match type 'struct VP_SURFACE *'
  568 | typedef struct _VPHAL_VEBOX_SURFACE_STATE_CMD_PARAMS
      | 
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o: in function `DecodeJpegPipelineAdapterXe_Lpm_Plus::~DecodeJpegPipelineAdapterXe_Lpm_Plus()':
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/codec/hal/dec/jpeg/pipeline/decode_jpeg_pipeline_adapter_xe_lpm_plus_base.h:41: undefined reference to `vtable for DecodeJpegPipelineAdapterXe_Lpm_Plus_Base'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o: in function `DecodeJpegPipelineAdapterXe_Lpm_Plus::~DecodeJpegPipelineAdapterXe_Lpm_Plus()':
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/codec/hal/dec/jpeg/pipeline/decode_jpeg_pipeline_adapter_xe_lpm_plus_base.h:41: undefined reference to `vtable for DecodeJpegPipelineAdapterXe_Lpm_Plus_Base'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x110): undefined reference to `typeinfo for DecodeJpegPipelineAdapterXe_Lpm_Plus_Base'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x4b8): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::Allocate(CodechalSetting*)'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x4c0): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::BeginFrame()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x4c8): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::EndFrame()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x4d0): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::Execute(void*)'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x4d8): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetStatusReport(void*, unsigned short)'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x4e0): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::Destroy()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x500): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::IsIncompletePicture()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x508): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::IsIncompleteJpegScan()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x510): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDummyReference()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x518): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDummyReferenceStatus()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x520): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::SetDummyReferenceStatus(_CODECHAL_DUMMY_REFERENCE_STATUS)'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x528): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetCompletedReport()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x530): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDecodeContext()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans109.ltrans.o:(.data.rel.ro+0x538): undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::GetDecodeContextHandle()'
/usr/bin/ld: /tmp/ccaCjPT0.ltrans119.ltrans.o: in function `CodechalInterfacesXe_Lpm_Plus::Initialize(void*, void*, MhwInterfacesNext*, _MOS_INTERFACE*)':
/builddir/build/BUILD/media-driver-intel-media-22.6.6/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus/codec/hal/dec/jpeg/decode_jpeg_pipeline_adapter_xe_lpm_plus.cpp:32: undefined reference to `DecodeJpegPipelineAdapterXe_Lpm_Plus_Base::DecodeJpegPipelineAdapterXe_Lpm_Plus_Base(CodechalHwInterfaceNext*, CodechalDebugInterface*)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Is #1595 necessary too, or something else?

Config flags used to build are:

```
  -DENABLE_NONFREE_KERNELS:BOOL=OFF \
  -DENABLE_KERNELS:BOOL=OFF \
  -DBUILD_KERNELS:BOOL=OFF \
  -DMEDIA_BUILD_FATAL_WARNINGS=OFF \
  -DBUILD_SHARED_LIBS:BOOL=OFF \
  -DGEN10:BOOL=OFF \
  -DVC1_Decode_Supported="no" \
  -DAVC_Encode_VME_Supported="no" \
  -DAVC_Encode_VDEnc_Supported="no" \
  -DHEVC_Encode_VME_Supported="no" \
  -DHEVC_Encode_VDEnc_Supported="no" \
  -DAVC_Decode_Supported="no" \
  -DHEVC_Decode_Supported="no" \
  -DAVC_Decode_Supported="no" \
  -G Ninja
```

frantisekz, 2023-01-13:
Hmm, adding -DJPEG_Decode_Supported="no" works around the issue, it's not ideal, but definitely not the end of the world. Shall I create a separate issue for this one?

bai-isaac, 2023-01-16:
> Hmm, adding -DJPEG_Decode_Supported="no" works around the issue, it's not ideal, but definitely not the end of the world. Shall I create a separate issue for this one?

Thanks frantisekz for pointing out this issue! Please create a issue, I'll take a look about this.  Thanks!

frantisekz, 2023-01-16:
> > Hmm, adding -DJPEG_Decode_Supported="no" works around the issue, it's not ideal, but definitely not the end of the world. Shall I create a separate issue for this one?
> 
> Thanks frantisekz for pointing out this issue! We will take a look about this. Thanks!

Thanks a lot, I've confirmed that your fix fixes the issue!

Closing this then.

closed 2023-01-16 by frantisekz