intel/media-driver #1541: [Bug]: Hardware video acceleration not working properly

pojlFDlxCOvZ4Kg8y1l4, 2022-11-04:
### Which component impacted?

Decode, Not sure

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Using HEVC hardware transcoding with jellyfin results in a bad picture. 
The CPU is a Intel Celeron J4105. 

Using software transcoding, direct play works fine. 
Also VP8, VP9 and x264 hardware transcoding results in good picture with a little higher than expected CPU utilization.  

Examples of the bad quality:

[image]  
https://user-images.githubusercontent.com/42654642/199727380-ec9bc92e-0c1d-4f0f-8c98-f780949258a0.mp4 

The package maintainer thinks the drivers are problematic, which are also a guess from the jellyfin dev.  
[Link to the issue with the package maaintainer](https://github.com/alexbelgium/hassio-addons/issues/537)
[Link to the issue with the jellyfin github](https://github.com/jellyfin/jellyfin/issues/8683)

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

1.  docker exec fb34bd1df5ee /usr/lib/jellyfin-ffmpeg/vainfo

libva info: VA-API version 1.16.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_16
libva info: va_openDriver() returns 0
Trying display: drm
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.5.4 (e461c05)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD

2. docker exec fb34bd1df5ee ls /dev/dri
by-path
card0
renderD128

 3 provided in step 1

4 does not work in the add-on, or I'm doing something wrong

### Do you want to contribute a patch to fix the issue?

No.

intel-mediadev, 2022-11-09:
Auto Created VSMGWL-58794 for further analysis.

XinfengZhang, 2022-11-16:
looks it is an hevc encode issure? the artifacts is exist in output bitstream playback?,

XinfengZhang, 2022-11-17:
it is GLK , https://ark.intel.com/content/www/us/en/ark/products/128989/intel-celeron-j4105-processor-4m-cache-up-to-2-50-ghz.html

pojlFDlxCOvZ4Kg8y1l4, 2022-11-17:
> looks it is an hevc encode issure? the artifacts is exist in output bitstream playback?,

I do not have a display attached, the weird output happens with different drivers.  
There was a change in behavior with a different driver in the alexbelgium jellyfin add-on.  

I've also run commands which other "jellyfin" users/devs mention to have no problems with.  

I would like to refer to these two issues I've created at both alexbelgiums repo (jellyfin on home assistant add-on) which bases off the linuxserver.io add-on, which in turn uses jellyfin from the jellyfin repo.  
These people have tried to troubleshoot me, there's also multiple commands listed which I've tried but did not solve the issue.  

[jellyfin repo issue](https://github.com/jellyfin/jellyfin/issues/8683)
[alexbelgium jellyfin issue](https://github.com/alexbelgium/hassio-addons/issues/537)

Do you need further information or maybe clarifications from the other issues?