intel/media-driver #1542: [Bug]:

heyu1, 2022-11-07:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

In Ubuntu22.04, when build "media_driver"/master (988c748aeee57), met below error:

===build info=====:
[ 48%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_VP.dir/__/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus/vp/hal/pipeline/vp_pipeline_adapter_xe_lpm_plus.cpp.o
In file included from repo/media-driver/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/vp/hal/feature_manager/vp_feature_manager_xe_lpm_plus_base.cpp:30:
repo/media-driver/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/hw/mhw_vebox_xe_lpm_plus_base_next_impl.h: In member function ‘virtual MOS_STATUS mhw::vebox::xe_lpm_plus_next::Impl::AddVeboxTilingConvert(PMOS_COMMAND_BUFFER, PMHW_VEBOX_SURFACE_PARAMS, PMHW_VEBOX_SURFACE_PARAMS)’:
repo/media-driver/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/hw/mhw_vebox_xe_lpm_plus_base_next_impl.h:2714:22: error: ‘MEMORY_OBJECT_CONTROL_STATE’ {aka ‘union MEMORY_OBJECT_CONTROL_STATE_REC’} has no member named ‘XE_LPG’; did you mean ‘XE_HP’?
 2714 |                     .XE_LPG.Index;
      |                      ^~~~~~
      |                      XE_HP

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

Jexu, 2022-11-08:
This maybe caused by incorrect gmmlib version and could you update your installed gmmlib to latest version?

@MicroYY Does it need to add gmmlib version check for new files compiling?

heyu1, 2022-11-08:
Hi Jexu,
Before built media-driver, I've updated gmmlib to the latest : 
commit 416ee5042f9c192344650cf90d16594007dc72e1 (HEAD -> master, tag: intel-gmmlib-22.3.0, origin/master, origin/HEAD).

Jexu, 2022-11-08:
You probably check if it is installed into correct place, I can reproduce above issue with old gmmlib, but it is gone when updating to latest.

heyu1, 2022-11-08:
Thanks for the reminder, after re-installed gmmlib to my build machine with specified path '/usr', media-driver built successfully.

Jexu, 2022-11-11:
Close this issue now, free to re-open if having any other questions.

closed 2022-11-11 by Jexu