intel/media-driver #1547: [Feature]: hardware acceleration supported for Alder Lake i3-1215U integrated graphics

AustinTodd0307, 2022-11-16:
### What Feature?

I am not sure if this is the right place to ask:
Is hardware acceleration supported for the Integrated graphics on the Alder Lake i3-1215U chip?

I see listed support for ADL-S/P/N. Is this the definitive list of supported architectures?
Will future releases provide support, or is hardware acceleration even possible for this chipset?

I am attempting (unsuccessfully) to use the latest intel-media-driver release (22.5.4) with the following dependencies:
- libva 2.16.0
- intel-gmmlib-22.2.0

Or will support be provided in oneVPL?

### What's the usage scenario would be benifited?

Others

### What impacted?

I am attempting to get video acceleration working on a NUC12 equip with the i3-1215U chip.

XinfengZhang, 2022-11-17:
it should already supported if you are talking about 
https://ark.intel.com/content/www/us/en/ark/products/226263/intel-core-i31215u-processor-10m-cache-up-to-4-40-ghz.html
the code here:
https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_sysinfo_g12.cpp#L471
shows it is from last year

if no other question, could you help to close this issue?

AustinTodd0307, 2022-11-17:
Thank you much.

closed 2022-11-17 by AustinTodd0307