intel/media-driver #1548: [Regression]: ffmpeg-vaapi decode Vp910bit/hevc10bit/hevc a large number of case failed on DG2

wenqingx, 2022-11-17:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

reproduce env:
libva:https://github.com/intel/libva/commit/ed7b12d131fbfaaebb91361e57e44c393897ec39
libva-utils:https://github.com/intel/libva-utils/commit/de2bc6821673a0708294f0e5e0452e6094db6762
gmmlib:https://github.com/intel/gmmlib/commit/e7d5cb083f6ff3a606d056303f0f7802e97cbca7
good media-driver:https://github.com/intel/media-driver/commit/2b3c3b315f9f920bf0d2a15bbe768b0d508e8185
first bad commit:https://github.com/intel/media-driver/commit/d4b51b5e133c60742e3ec43012bac6aa88610969

After apply  https://github.com/intel/media-driver/commit/d4b51b5e133c60742e3ec43012bac6aa88610969, a large number of failure cases will occur in decode 10bithevc/10bitvp9/hevc.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

High priority, please take priority. This involves DG2 PV

intel-mediadev, 2022-11-17:
Auto Created VSMGWL-59094 for further analysis.

wenqingx, 2022-11-17:
An example cmdline:
ffmpeg -v verbose -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format p010le -hwaccel_flags allow_profile_mismatch  -i input.ivf -lavfi 'null' -c:v rawvideo -pix_fmt p010le -fps_mode passthrough -autoscale 0 -vframes 17 -y -f tee 'output.yuv|[f=md5]pipe:1'

log:
AVHWDeviceContext @ 0x55dd1726e180] libva: VA-API version 1.17.0
[AVHWDeviceContext @ 0x55dd1726e180] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x55dd1726e180] libva: Trying to open /home/media-ci/media-env/media/build/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55dd1726e180] libva: Found init function __vaDriverInit_1_17
[AVHWDeviceContext @ 0x55dd1726e180] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x55dd1726e180] Initialised VAAPI connection: version 1.17
[AVHWDeviceContext @ 0x55dd1726e180] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 22.6.0 (d4b51b5e1).
[AVHWDeviceContext @ 0x55dd1726e180] Driver not found in known nonstandard list, using standard behaviour.
Selecting decoder 'vp9' because of requested hwaccel method vaapi
  Duration: 00:00:00.02, start: 0.001000, bitrate: 195458 kb/s
  Stream #0:0: Video: vp9 (Profile 2), 1 reference frame (VP90 / 0x30395056), yuv420p10le(pc, bt470bg/unknown/unknown), 432x160, 1k tbr, 1k tbn
Stream mapping:
  Stream #0:0 (vp9) -> null:default
  null:default -> Stream #0:0 (rawvideo)
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x55dd1747d300] w:432 h:160 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[tee @ 0x55dd173698c0] filename:'output.yuv' format:rawvideo
[tee @ 0x55dd173698c0]     stream:0 codec:rawvideo type:video bsfs: null
[tee @ 0x55dd173698c0] filename:'pipe:1' format:md5
[tee @ 0x55dd173698c0]     stream:0 codec:rawvideo type:video bsfs: null
Output #0, tee, to 'output.yuv|[f=md5]pipe:1':
  Metadata:
    encoder         : Lavf59.34.101
  Stream #0:0: Video: rawvideo, 1 reference frame, p010le(pc, bt470bg/unknown/unknown, progressive), 432x160 (0x0), q=2-31, 1036800 kb/s, 1k fps, 1k tbn
    Metadata:
      encoder         : Lavc59.50.100 rawvideo
[graph 0 input from stream 0:0 @ 0x55dd173742c0] w:376 h:392 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[graph 0 input from stream 0:0 @ 0x55dd17374b80] w:408 h:328 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[graph 0 input from stream 0:0 @ 0x55dd17373380] w:471 h:502 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[graph 0 input from stream 0:0 @ 0x55dd173eb540] w:304 h:365 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[graph 0 input from stream 0:0 @ 0x55dd17377ac0] w:471 h:502 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[graph 0 input from stream 0:0 @ 0x55dd173ddd80] w:396 h:368 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[graph 0 input from stream 0:0 @ 0x55dd173ecf00] w:144 h:264 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[tee @ 0x55dd173698c0] Non-monotonous DTS in output stream 0:0; previous: 9, current: 0; changing to 10. This may result in incorrect timestamps in the output file.
[graph 0 input from stream 0:0 @ 0x55dd173ddd80] w:327 h:328 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[tee @ 0x55dd173698c0] Non-monotonous DTS in output stream 0:0; previous: 10, current: 1; changing to 11. This may result in incorrect timestamps in the output file.
[tee @ 0x55dd173698c0] Non-monotonous DTS in output stream 0:0; previous: 11, current: 3; changing to 12. This may result in incorrect timestamps in the output file.
[graph 0 input from stream 0:0 @ 0x55dd173eb540] w:299 h:328 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[tee @ 0x55dd173698c0] Non-monotonous DTS in output stream 0:0; previous: 12, current: 4; changing to 13. This may result in incorrect timestamps in the output file.
[graph 0 input from stream 0:0 @ 0x55dd173eb540] w:432 h:488 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[tee @ 0x55dd173698c0] Non-monotonous DTS in output stream 0:0; previous: 13, current: 5; changing to 14. This may result in incorrect timestamps in the output file.
[graph 0 input from stream 0:0 @ 0x55dd1747f340] w:368 h:506 pixfmt:p010le tb:1/1000 fr:1000/1 sar:0/1
[AVHWFramesContext @ 0x7fbbac03e780] Failed to sync surface 0x187: 1 (operation failed).
[vp9 @ 0x55dd1736a700] Failed to transfer data to output frame: -5.
Error while processing the decoded data for stream #0:0
[tee @ 0x55dd173698c0] Non-monotonous DTS in output stream 0:0; previous: 14, current: 6; changing to 15. This may result in incorrect timestamps in the output file.
All streams finished for output file #0
Terminating muxer thread 0
[AVIOContext @ 0x55dd17291240] Statistics: 488646 bytes read, 0 seeks
Conversion failed!

except:pass

Jexu, 2022-11-17:
It is caused by unreached surface buffer since the commit changed behavior of buffer heap loop.
I will revert this commit later soon while this coomit is not expected.

Jexu, 2022-11-18:
@wenqingx could you try this commit https://github.com/intel/media-driver/commit/5d30f0bceaff477257667ed560ae28aa7f6a4bdd

wenqingx, 2022-11-21:
> @wenqingx could you try this commit [5d30f0b](https://github.com/intel/media-driver/commit/5d30f0bceaff477257667ed560ae28aa7f6a4bdd)

Is this just revert https://github.com/intel/media-driver/commit/d4b51b5e133c60742e3ec43012bac6aa88610969 ? I tried  revert it manually locally, and the issue can be fixed.

Jexu, 2022-11-22:
Ok, since the revert commit has been merged, so i will close this issue now.

closed 2022-11-22 by Jexu

wenqingx, 2022-11-22:
thanks for your help.