intel/media-driver #1549: [Bug]: va/va_dricommon.h: No such file or directory

heitbaum, 2022-11-18:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Should have compiled against Libva 2.16.0

```
$ find . -name va_dricommon.h
./build.LibreELEC-Generic.x86_64-11.0-devel/build/libva-2.16.0/va/x11/va_dricommon.h
```

```
BUILD      media-driver (target)
    TOOLCHAIN      cmake (auto-detect)
Executing (target): cmake -GNinja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/etc/cmake-x86_64-libreelec-linux-gnu.conf -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -DBUILD_CMRTLIB=OFF -DBUILD_KERNELS=ON -DBUILD_TYPE=release -DENABLE_NONFREE_KERNELS=ON -DMEDIA_BUILD_FATAL_WARNINGS=ON -DMEDIA_RUN_TEST_SUITE=OFF /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.3
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_INSTALL_PREFIX = /usr
-- Found PkgConfig: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/pkg-config (found version "0.29.2") 
-- Could NOT find X11 (missing: X11_X11_INCLUDE_PATH X11_X11_LIB) 
-- Looking for backtrace
-- Looking for backtrace - found
-- backtrace facility detected in default set of libraries
-- Found Backtrace: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include  
-- Checking for module 'igdgmm>=12.0.0'
--   Found igdgmm, version 12.3.0
-- media -- PLATFORM = linux
-- media -- ARCH = 64
-- media -- CMAKE_CURRENT_LIST_DIR = /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.3/media_driver
-- media -- INCLUDED_LIBS = 
-- media -- LIB_NAME = iHD_drv_video
-- media -- OUTPUT_NAME = 
-- media -- BUILD_TYPE/UFO_BUILD_TYPE/CMAKE_BUILD_TYPE = release/release/Release
-- media -- LIBVA_INSTALL_PATH = 
-- media -- MEDIA_VERSION = 22.6.3
-- Checking for module 'libva>=1.8.0'
--   Found libva, version 1.16.0
-- LIBVA_DRIVERS_PATH = /usr/lib/dri
-- Configuring done
-- Generating done
-- Build files have been written to: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.3/.x86_64-libreelec-linux-gnu
Executing (target): ninja 

…

[335/1467] Building CXX object media_driver/CMakeFiles/iHD_drv_v....dir/__/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp.o
FAILED: media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/__/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp.o 

…

/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.3/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:26:10: fatal error: va/va_dricommon.h: No such file or directory
   26 | #include <va/va_dricommon.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
[350/1467] Building CXX object media_driver/CMakeFiles/iHD_drv_v.../media_softlet/linux/common/ddi/media_libva_interface_next.cpp.o
ninja: build stopped: subcommand failed.
```

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

with commit https://github.com/intel/media-driver/commit/1e2afc600f35d35cd2790287516541bcbd32de61 compile fails with `media-driver-22.6.3/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:26:10: fatal error: va/va_dricommon.h: No such file or directory `

https://github.com/intel/media-driver/blob/0ba2777c8565d66065da53ea2c375b17fb4a066a/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp#L26

### Debug Information

gmmlib: update to 22.3.1
media-driver: update to 22.6.3
libva: update to 2.16.0

### Do you want to contribute a patch to fix the issue?

No.

Jexu, 2022-11-22:
Do you try to install these new dependencies  before build libva.
https://github.com/intel/media-driver/pull/1509/files

heitbaum, 2022-11-22:
> Do you try to install these new dependencies before build libva. https://github.com/intel/media-driver/pull/1509/files

Hi @jexu - the file is `va/x11/va_dricommon.h` which is from libva. The build we are doing in not an X11 build (but a GBM build). 

In these that file is not there (and it is only as `/va/x11/` from libva.
- https://packages.ubuntu.com/bionic/amd64/libx11-xcb-dev/filelist
- https://packages.ubuntu.com/bionic/amd64/libxcb-dri3-dev/filelist

Jexu, 2022-11-22:
I know it is from libva. Without above dependencies, you may unable to install the header file when build libva.
Can you share autogen.sh output log for libva

heitbaum, 2022-11-22:
Hi @Jexu - below is the `libva` we have been building. I updated the package depends to include `libxcb` - to match your suggested - but still no `va/va_dricommon.h`.

https://github.com/LibreELEC/LibreELEC.tv/blob/5132be1a1d7667c5d601e8e18cd7ed695e6e9ac2/packages/multimedia/libva/package.mk#L20-L28

```
-  PKG_DEPENDS_TARGET="toolchain libdrm"
+  PKG_DEPENDS_TARGET="toolchain libdrm libxcb"
```

### meson.log - without the addition of `libxcb`
```
libva-2.16.0/.x86_64-libreelec-linux-gnu/meson-logs$ cat meson-log.txt 
Build started at 2022-11-22T09:09:34.199142
Main binary: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/python3
Build Options: -Dstrip=true -Ddisable_drm=false -Denable_docs=false -Denable_va_messaging=true -Dwith_x11=no -Dwith_glx=no -Dwith_wayland=no -Dprefix=/usr -Dbindir=/usr/bin -Dlibdir=/usr/lib -Dlibexecdir=/usr/lib -Dlocalstatedir=/var -Dsbindir=/usr/sbin -Dsysconfdir=/etc -Dbuildtype=plain '--cross-file /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/libva-2.16.0/.x86_64-libreelec-linux-gnu/meson.conf'
Python system: Linux
The Meson build system
Version: 0.64.0
Source dir: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/libva-2.16.0
Build dir: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/libva-2.16.0/.x86_64-libreelec-linux-gnu
Build type: cross build
Project name: libva
Project version: 2.16.0
-----
Detecting compiler via: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc --version
compiler returned <Popen: returncode: 0 args: ['/var/media/DATA/home-rudi/LibreELEC.tv/build.L...>
compiler stdout:
x86_64-libreelec-linux-gnu-gcc-12.2.0 (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

compiler stderr:

Running command: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc -E -dM -
-----
Detecting linker via: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc -Wl,--version -Wl,--as-needed -fuse-ld=gold
linker returned <Popen: returncode: 0 args: ['/var/media/DATA/home-rudi/LibreELEC.tv/build.L...>
linker stdout:
GNU gold (GNU Binutils 2.39) 1.16
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

linker stderr:
collect2 version 12.2.0
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/12.2.0/../../../../x86_64-libreelec-linux-gnu/bin/ld.gold -plugin /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/12.2.0/liblto_plugin.so -plugin-opt=/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/12.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccnbVESR.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --sysroot=/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot --eh-frame-hdr --hash-style=gnu -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/crt1.o /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/crti.o /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/12.2.0/crtbegin.o -L/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/12.2.0 -L/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/12.2.0/../../../../x86_64-libreelec-linux-gnu/lib -L/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib --version --as-needed -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/12.2.0/crtend.o /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/crtn.o

Sanity testing C compiler: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc
Is cross compiler: True.
Sanity check compiler command line: /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc sanitycheckc.c -o sanitycheckc_cross.exe -march=x86-64-v3 -Wall -pipe -O2 -fomit-frame-pointer -DNDEBUG [... cut ...]
    libdir             : /usr/lib
    libexecdir         : /usr/lib
    localstatedir      : /var
    prefix             : /usr
    sbindir            : /usr/sbin
    strip              : true
    sysconfdir         : /etc
    disable_drm        : false
    enable_docs        : false
    enable_va_messaging: true
    with_glx           : no
    with_wayland       : no
    with_x11           : no

Found ninja-1.11.1 at /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/ninja
```

heitbaum, 2022-11-22:
In the meson.build the `x11/va_ dricommon.h` is within the `if WITH_X11`
https://github.com/intel/libva/blob/393e7af16d4787034df03587191e4c245f853482/va/meson.build#L154

the Makefile.am looks like it might behave differently to the meson.build - as it has `libva_x11includedir` =`${includedir}/va`
https://github.com/intel/libva/blob/393e7af16d4787034df03587191e4c245f853482/va/x11/Makefile.am#L41-L56

```
libva-2.16.0$ find . -type f -exec grep -H va_dricommon.h {} \;
./va/x11/dri2_util.c:#include "va_dricommon.h"
./va/x11/va_dricommon.c:#include "va_dricommon.h"
./va/x11/va_x11.c:#include "va_dricommon.h"
./va/x11/Makefile.am:   va_dricommon.h          \
./va/meson.build:    'x11/va_dricommon.h',
```

```
media-driver-22.6.3$ find . -type f -exec grep -H va_dricommon.h {} \;
./media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:#include <va/va_dricommon.h>
./media_driver/linux/common/ddi/media_libva_putsurface_linux.h:#include <va/va_dricommon.h>
```

heitbaum, 2022-11-22:
More testing:

```
cp build.LibreELEC-Generic.x86_64-11.0-devel/build/libva-2.16.0/va/x11/va_dricommon.h \
   build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/va
…
## compiled now fails with `X11/Xlib.h`
- so looking like x11/va_dricommon.h is not appropriate for non X11 builds
…
In file included from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.6.3/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:26:
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/va/va_dricommon.h:28:10: fatal error: X11/Xlib.h: No such file or directory
   28 | #include <X11/Xlib.h>
      |          ^~~~~~~~~~~~
compilation terminated.
```
So is the correct fix as below?

```
#if defined(X11_FOUND)
#include <va/va_dricommon.h>
#endif 
```

Jexu, 2022-11-23:
Yeah, the header should be used with macro definition since no x11 found.

Nexarian, 2022-11-26:
This is blocking the build for XRDP's prototype of Intel media driver hardware acceleration. I can enable X11 for the LibVA build as a temporary workaround, but I'd like this fixed as soon as possible. Do you want me to cut a PR to fix it?

PR #1550 (2022-11-28): [Media Common] [VP] Remove unused include fixes #1549
[Media Common] [VP] dont include va_dricommon.h on non X11 builds

<va/va_dricommon.h> is unavailable and not in non-X11 libva 2.16.0

- Fixes #1549 

tested with GBM build.
closed 2022-11-28 by intel-mediadev