intel/media-driver #1551: [Bug]: Fails to build on gcc-4.9 (seems specific to GEN12)

th0ma7, 2022-11-19:
### Which component impacted?

Not sure

### What happened?

Synocommunity release of the driver: https://github.com/SynoCommunity/spksrc/pull/5495
Using latest official: [22.5.4](https://github.com/intel/media-driver/releases/tag/intel-media-22.5.4)

Able to circumvent the build error using conjunction of the following cmake options:
```
#CMAKE_ARGS += -DGEN11=OFF
#CMAKE_ARGS += -DGEN12=OFF
CMAKE_ARGS += -DDG2=OFF
CMAKE_ARGS += -DPVC=OFF
CMAKE_ARGS += -DXE_LPG=OFF
CMAKE_ARGS += -DXEHP_SDV=OFF
CMAKE_ARGS += -DXe_M_plus=OFF
CMAKE_ARGS += -DMTL=OFF
```

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

n/a

### Debug Information

Build error output:
[build-x64-6.1.log](https://github.com/intel/media-driver/files/10047900/build-x64-6.1.log)

dvrogozh, 2022-11-21:
Please, check top level readme of media driver: https://github.com/intel/media-driver#known-issues-and-limitations

Quoting: "Intel(R) Media Driver for VAAPI is recommended to be built against gcc compiler v6.1 or later, which officially supported C++11."

media-driver (and VPL RT and MSDK) should be built using gcc version 6 or later. Earlier versions do not fully and officially support C++11 (they have only experimental support). Moreover, media driver is known to have functional issues if built with earlier versions of gcc:. Please, consider to update your compiler per media-driver build requirements.

As per above this is not a bug if not reproduced on gcc version 6 or later. Can this issue be reproduced on these gcc versions?

LhGu, 2022-11-30:
mhw_vdbox_avp_impl.h  compile error

Jexu, 2022-12-05:
As @dvrogozh mentioned, media driver introduces C++11 which requires newer version of gcc support. So could you try to upgrade gcc to check if it is gone?

th0ma7, 2022-12-05:
There are two things to consider:
1. I cannot upgrade the gcc compiler suite as I'm using Synology's default toolchain that comes with DSM 6.x release (they did upgrade to gcc-7.5 for DSM 7.x but that cannot be used to compile for older OS release)
2. gcc-4.9 does support c++11 while c++1y (e.g. c++14) was partial.

dvrogozh, 2022-12-05:
> gcc-4.9 does support c++11 

No, it has only experimental support for C++11. Read documentation here: https://gcc.gnu.org/onlinedocs/gcc-4.9.4/gcc/Standards.html#Standards. Quoting (C++ section): "GCC supports the original ISO C++ standard (1998) and contains **experimental support for the second ISO C++ standard (2011)**." Same for gcc-5. Only gcc-6 has declared non-experimental support for c++11. In practice "experimental" support means that if you will find a compiler bug in gcc-4.9 for c++11 it won't be fixed in this compiler version.

The chance you have to proceed with this issue is to either reproduce same issue you have with other compiler which officially supports c++11 (for gcc that's gcc-6 or later), or convince that a faulty statement does not follow the standard despite compilers with official support can handle it, or to suggest a refactor for the code which will simplify the place out.

> they did upgrade to gcc-7.5 for DSM 7.x but that cannot be used to compile for older OS release

Consider to switch to newer release. It's better than make something buildable with unsupported compiler version and step later in non-fixable compiler bugs which would show as runtime errors or segmentation faults.

bkuhls, 2022-12-05:
> Moreover, media driver is known to have functional issues if built with earlier versions of gcc:. Please, consider to update your compiler per media-driver build requirements.

Hi,

sitting in the same boat with a gcc-4.9 system which, due to various reasons, I can not update I accept your recommendation but for my use case (Kodi decoding videos with a gen12 cpu) mediadriver-2.6.3 it works flawlessly.

To accomplish this I am patching mediadriver and use some configure options to disable problematic parts:
https://github.com/bkuhls/buildroot/commit/fe71590a0b7e1e4d6b882d65cc249234f202aeef

patch 0002-gcc49.patch is available as https://github.com/intel/media-driver/pull/1545
patch 0004-gcc49.patch is an ugly hack ;)

Jexu, 2022-12-06:
Fix similar with #1545 could temporarily solve the issue, but i think it is not necessary. Media dirver has allowed developpers to use newer c++, especially in newly released driver. It cannot guarantee similar build issues by older compiler will not exist again in the future.
So, if you really cannot upgrade the compiler, manual patch may help.

closed 2022-12-20 by Jexu