intel/media-driver #1553: [Bug]:  error: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'uint64_t' {aka 'long long unsigned int'} [-Werror=format=]

SuperSandro2000, 2022-11-21:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

After updating to 2.6.3 building on i686-linux fails with the following error:

```
/build/source/media_softlet/linux/common/ddi/media_libva_util_next.cpp: In static member function 'static void MediaLibvaUtilNext::MediaPrintFps()':
/build/source/media_softlet/linux/common/ddi/media_libva_util_next.cpp:2199:47: error: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'uint64_t' {aka 'long long unsigned int'} [-Werror=format=]
 2199 |         sprintf(temp,"FPS:%6.4f, Interval:%11lu\n", fps,((uint64_t)tv2.tv_sec)*1000 + (tv2.tv_usec/1000));
      |                                           ~~~~^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |                                     |
      |                                               long unsigned int                     uint64_t {aka long long unsigned int}
```

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

I am maintaining the intel-media-driver package for NixOS and the 32 bit variant is required for hardware acceleration in combination with Steam.

### Do you want to contribute a patch to fix the issue?

No.

intel-mediadev, 2022-11-22:
Auto Created VSMGWL-59303 for further analysis.

cqian2, 2022-11-23:
"%lu" is for unsigned long, in current gcc (gcc12) 32bit compiler, unsigned long takes 4 bytes, while unsigned long long takes 8 bytes, 8 bytes -> 4 bytes has the data lossing risk, though number "11" was added in "%11lu" as the original purpose was only low11 bits data were cared (needed to be printed), but the compiler would not recognize this. 
While in current gcc (gcc12) 64bit compiler, unsigned long takes 8 bytes, 8 bytes -> 8 bytes has no risk.
Whether or not "11" is added has no impact on the compiler behavior.

cqian2, 2022-11-23:
@SuperSandro2000 , would you please help verify this PR: https://github.com/intel/media-driver/pull/1557 ?

SuperSandro2000, 2022-11-26:
That's beyond my knowledge. 😓

cqian2, 2022-11-28:
Thanks for the verification, @SuperSandro2000, would you please help close this issue then?

SuperSandro2000, 2022-11-28:
Just change the comment in the PR to ``Fixes #1553`` then it gets auto closed when the PR is merged.

cqian2, 2022-11-28:
The PR#1557 got merged, close the issue now.

closed 2022-11-28 by cqian2