intel/media-driver #1554: [Bug]: Blur occured during non videowall case composition on EHL

james-tang17, 2022-11-21:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. In Ubuntu 20.04.3 with default kernel 5.15.0
2. Media SDK: 21.1.3 / libva: 2.11.0 / media-driver: 21.1.3 / gmmlib: 21.1.1
3. Run media sdk sample_multi_transcode with the following par file, with 4x4 layout but one of the output of these input stream is smaller than others which will lead the compostion flow in MSDK to go with Execute_Composition path. 
4. the result will show blur on all input streams beside the last one. This issue can only be reproudced on EHL, no issue on TGL-U with the same environment.

par file:
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 0 -vpp_comp_dst_y 0 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 480 -vpp_comp_dst_y 0 -vpp_comp_dst_w 320 -vpp_comp_dst_h 240 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 960 -vpp_comp_dst_y 0 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 1440 -vpp_comp_dst_y 0 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30

-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 0 -vpp_comp_dst_y 270 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 480 -vpp_comp_dst_y 270 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 960 -vpp_comp_dst_y 270 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 1440 -vpp_comp_dst_y 270 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30

-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 0 -vpp_comp_dst_y 540 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 480 -vpp_comp_dst_y 540 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 960 -vpp_comp_dst_y 540 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 1440 -vpp_comp_dst_y 540 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30

-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 0 -vpp_comp_dst_y 810 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 480 -vpp_comp_dst_y 810 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 960 -vpp_comp_dst_y 810 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30
-i::h264 ./video/1080p.h264 -join -hw -async 4   -o::sink -vpp_comp_dst_x 1440 -vpp_comp_dst_y 810 -vpp_comp_dst_w 480 -vpp_comp_dst_h 270 -ext_allocator -fps 30

-vpp_comp_only 16 -w 1920 -h 1080 -async 4 -threads 2 -join -hw -i::source -ext_allocator -vpp::vid -fps 30 -ec::rgb4 -rdrm

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

customer need to see all the image are not in blur.

### Debug Information

[vp_dump-not-video-wall-ehl.tar.gz](https://github.com/intel/media-driver/files/10054871/vp_dump-not-video-wall-ehl.tar.gz)
this is the vp_dump log.

FurongZhang, 2022-11-21:
@MicroYY  to take a look.

intel-mediadev, 2022-11-22:
Auto Created VSMGWL-59302 for further analysis.

MicroYY, 2022-11-25:
A potential fix: https://github.com/intel/media-driver/pull/1561