intel/media-driver #1558: [Bug]: Alder Lake HW video decoder outputs corrupted video when not connected to a display

phaserblast, 2022-11-24:
### Which component impacted?

Decode

### What happened?

I am having an issue where if I try to use the on-chip Alder Lake i5-12400 hardware decoder via VAAPI, it only works if a DisplayPort monitor is actually connected to the CPU's video output. I am using EGLImages to store the decoded VAAPI frames then rendering via OpenGL. If I try to use the CPU's decoder while another graphics card is running the display, the output from the CPU's video decoder is corrupted. This is not an issue with my AMD card also installed, which will happily decode video to EGLImages which can be displayed on any EGL context running on another graphics card.

Summary:

- Decode on CPU with display connected to CPU's video output, render OpenGL on CPU-connected display - **OK**
- Decode on AMD PCIe graphics card (no display connected), render OpenGL with CPU graphics - **OK**
- Decode on CPU (no display connected), render OpenGL on a PCIe graphics card - **BROKEN**

I am running Ubuntu with the 6.1 rc5 kernel, latest Intel media driver (non-free) git and also latest mesa git (as of yesterday).
The expected behavior is I should be able to decode video on the CPU, then use either `vaDeriveImage` or `vaExportSurfaceHandle` from the CPU wrapped in EGLImage to render via OpenGL on another video card. The cross-GPU solution only works when I decode on the AMD device.

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

/dev/dri contains three entries:

1. [card0] Built-in Alder Lake GPU (i5-12400)
2. [card1] Intel Arc A750
3. [card2] AMD Radeon RX 640

Jexu, 2022-11-25:
>>>Decode on CPU (no display connected), render OpenGL on a PCIe graphics card - BROKEN

Do you check if the deocde output from iGPU(on CPU) is good or corrupted?  Here is a little suspicious that we may need to confirm it is iGPU decode issue or OpenGL rendering causing corrupted.

Jexu, 2022-11-28:
When you get the VAImage, this frame decoding is finished, with no corrupted since VAImage output is good. But i am not very sure if coverting from VAImage to EGLImage causing some problem or rendering EGImage causing corrupted. These two processes are out of media driver.
Could you try the experiment by decoding next frame only after previous one rendering completed. This could verify if iGPU overwrites rendering surface.

Jexu, 2023-01-17:
So far, may I know if your issue is solved?

phaserblast, 2023-01-17:
> So far, may I know if your issue is solved?

Unfortunately, no. The problem is still there on kernel 6.2 and Mesa 23.1.0-devel. I created an issue on GitLab for Mesa, but nobody seems interested in investigating the bug. I suspect it is a problem with the Intel iris driver, as it doesn't seem to affect AMD GPUs:

https://gitlab.freedesktop.org/mesa/mesa/-/issues/7779

I made VAImage dumps from both the Alder Lake iGPU and Arc A750 decoders, and the images seem fine. Only EGLImages are corrupted.

Jexu, 2023-07-12:
Any update? I saw you have some update in mesa issue that maybe related to buffer tiling.
Do you check tile format are aligned between exported VAImage and imported EGLImage?

Jexu, 2023-09-25:
Let me close this issue firstly for no respose for long time while it trends to issue in EGLImage. Please fell free to repoen it again if having any concerns.

closed 2023-09-25 by Jexu