intel/media-driver #1564: [Bug]: USERPTR apparently results in tiled surface

dwlsalmeida, 2022-11-28:
### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

I have attached two sets of VA traces and their respective surfaces, which I have dumped through `LIBVA_TRACE_SURFACE="dec.yuv"`

You can see that both traces are identical, save for the fact that one imports a userspace allocation through userptr while creating the surface.

The surface backed by VA memory decodes fine, as can be seen in the surface dump, but the userptr one does not.

Zooming into 16x240 chunks on the luma plane, one can see a pattern where it appears that the surface contains tiled data: I've also attached a screenshot from YUView to better explain what I mean.

This appears to be a bug within the Intel driver, am I correct?

All dumped data can be found in the following file: 
[dump.zip](https://github.com/intel/media-driver/files/10104704/dump.zip)

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

So far, H264 decoding with userptr.

### Debug Information

vainfo is attached

XinfengZhang, 2022-12-06:
userptr is linear , userptr use DRM_IOCTL_I915_GEM_USERPTR, then it was treated as a tiling surfaces, becaue HW decode only support tiling output.

dwlsalmeida, 2022-12-06:
Not sure I understand. Is there a way to get linear output from userptr?

XinfengZhang, 2023-05-06:
sorry for late response, but the userptr itself is linear, what I am saying is:
1. how to use the userptr surfaces? , it could not be decoded output (decoded HW only accept tiling surfaces, so if it work, it will treat it as tiling. 
2. which platform are you using , the get_tiling function maybe is invalid in some platform, may cause potential issue.?

dwlsalmeida, 2023-06-13:
@XinfengZhang I am on Linux.

> how to use the userptr surfaces? , it could not be decoded output (decoded HW only accept tiling surfaces, so if it work, it will treat it as tiling.

Well, in my humble opinion, media-driver should either detile (as in vaGetImage) or refuse userptr for decoding. Right now, users believe they are getting non-tiled surfaces because media-driver uses DRM_IOCTL_I915_GEM_USERPTR. **The media-driver code also thinks that this surface is not tiled, so this is likely a bug**.

In DdiMediaUtil::AllocateSurface(), this code gets executed, for example:

```
            case I915_TILING_NONE: <------------
            default:
                gmmCustomParams.Flags.Info.Linear = true;
```

XinfengZhang, 2023-06-18:
so, you need a check , or temp surface to decoded and copy out to user ptr?

dwlsalmeida, 2023-06-22:
>  temp surface to decoded and copy out to user ptr?

Yes, that would be consistent with the rest of the code. As I said, everything in media-driver is written with the assumption that the USERPTR surface is linear, despite the hardware only being able to output in tiled formats, so a conversion is needed before the user can access IMHO.

dwlsalmeida, 2023-06-22:
@XinfengZhang

XinfengZhang, 2023-06-26:
TBH, looks we could use other method to handle this: 
method 1. a. create user ptr surfaces, b. decode, c. Call vaCopy to copy it to user ptr surface
method 2. a. create user ptr surfaces, b. deriveimage, c. decode, d. call vaGetImage to read the data from the surface to user ptr 
anyway, looks it need application changes. 

but if we only need user ptr as decode output , and it is transparent to application. need some effort. I am wondering the value

XinfengZhang, 2024-05-21:
because user ptr is alwasy linear one, so , need copy as previous comments, close it if no other concern

closed 2024-05-21 by XinfengZhang