intel/media-driver #1567: [Bug]: SynchronizeFirstTask return MFX_ERR_DEVICE_FAILED(-17), when enable lowpower params,have opened guc and huc

okfujian, 2022-12-02:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

OS: ubuntu 20.04

kernel:  Intel Linux kernel 5.10

MediaSDK 2022q2

CPU Model: 1165G7 Tiger-lake

We use the YUV data needed by the encoder is filled in by myself. mem type is SYSTEM_MEMORY.  -17  error occurs frequently in the coding process.

MFXVideoSession::virtual mfxStatus SyncOperation(mfxSyncPoint syncp, mfxU32 wait) return -17, and couldn't return to normal, unless destroy and recreate.

 

If only 265 or 264 is encoded,  the problem does not reappear temporarily.

If both h.265 and h.264 (with lowpower:on ) is encoded, the problem is appear in a few minutes.

 

Because we would use at least 8 stream encoding we must open lowpower parameters to make sure all encoding process is quick enough.

 

We have turned on GUC and HUC.

 

[10.12.32.197]# cat /sys/kernel/debug/dri/0/gt/uc/guc_info
GuC firmware: i915/tgl_guc_35.2.0.bin
status: RUNNING
version: wanted 35.2, found 35.2
uCode: 417344 bytes
RSA: 256 bytes

GuC status 0x8002f077:
Bootrom status = 0x3b
uKernel status = 0xf0
MIA Core status = 0x2

Scratch registers:
0: 0x0
1: 0x0
2: 0x7effd3
3: 0x0
4: 0x4000
5: 0x40
6: 0x1004
7: 0x0
8: 0x0
9: 0x0
10: 0x0
11: 0x0
12: 0x0
13: 0x0
14: 0x0
15: 0x0

GuC log relay not created
[TM1400 B640B6E7 ~ ]
[10.12.32.197]# cat /sys/kernel/debug/dri/0/gt/uc/huc_info
HuC firmware: i915/tgl_huc_7.5.0.bin
status: RUNNING
version: wanted 7.5, found 7.5
uCode: 580352 bytes
RSA: 256 bytes
HuC status: 0x00090001

 

 

/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h265 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &
/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h265 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &
/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h265 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &
/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h265 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &
 
/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h264 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 15000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &
/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h264 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 15000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &
/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h264 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 15000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &
/opt/intel/mediasdk/share/mfx/samples/_bin/sample_encode h264 -gpucopy::on -i /tmp/yuv/cnn.yuv -o test.h265 -nv12 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 15000 -fps 60 -f 60 -hw -async 2 -lowpower:on -u speed -timeout 600 &

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2022-12-02:
Auto Created VSMGWL-59581 for further analysis.

bai-isaac, 2023-06-16:
hi @okfujian , if tgl_huc_7.9.3.bin work well, can we close this issue?

closed 2023-06-16 by okfujian