intel/media-driver #1570: [Bug]:[MTL-P] encode with vah265lpenc failed.

UngTengEn, 2022-12-06:
### Is it regression? Good in old configuration?

None

### What happened?

1. Build libva 2.16 and latest media-driver (commit 33b5939ca)
2. Follow [here](https://github.com/intel-media-ci/cartwheel-gstreamer) to build the gstreamer VA module.
3. Decode and encode h265 video with "gst-launch-1.0 filesrc location=<h265 raw video> ! h265parse ! vah265dec ! vah265lpenc ! h265parse ! filesink location=Videos/test.h265" seeing this error -
[LIBVA]:CRITICAL - QueryVideoProcPipelineCaps:2640: Invalid ContextID.
0:00:00.024970568 393409 0x7f6ffb3f6890 ERROR               vafilter gstvafilter.c:335:gst_va_filter_ensure_pipeline_caps:<vafilter0> vaQueryVideoProcPipelineCaps: invalid VAContextID
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vah265lpenc0': gst.va.display.handle=context, gst-display=(GstObject)"\(GstVaDisplayDrm\)\ vadisplaydrm2", description=(string)"Intel\ iHD\ driver\ for\ Intel\(R\)\ Gen\ Graphics\ -\ 22.6.4\ \(9042c1745\)", path=(string)/dev/dri/renderD128;
Redistribute latency...
0:00:00.064073066 393409 0x7f6ffb2db980 ERROR              vaencoder gstvaencoder.c:384:gst_va_encoder_open:<vaencoder1> vaCreateConfig: attribute not supported
0:00:00.064083118 393409 0x7f6ffb2db980 ERROR              vah265enc gstvah265enc.c:4573:gst_va_h265_enc_reconfig:<vah265lpenc0> Failed to open the VA encoder.
0:00:00.064085600 393409 0x7f6ffb2db980 ERROR              vabaseenc gstvabaseenc.c:611:gst_va_base_enc_reset:<vah265lpenc0> Error at reconfiguration error
0:00:00.064125398 393409 0x7f6ffb2db980 ERROR              vaencoder gstvaencoder.c:384:gst_va_encoder_open:<vaencoder1> vaCreateConfig: attribute not supported
0:00:00.064127201 393409 0x7f6ffb2db980 ERROR              vah265enc gstvah265enc.c:4573:gst_va_h265_enc_reconfig:<vah265lpenc0> Failed to open the VA encoder.
0:00:00.064128728 393409 0x7f6ffb2db980 ERROR              vabaseenc gstvabaseenc.c:611:gst_va_base_enc_reset:<vah265lpenc0> Error at reconfiguration error
[LIBVA]:CRITICAL - MediaSurfaceToMosResource:66: nullptr mosResource->bo
[LIBVA]:CRITICAL - MediaSurfaceToMosResource:66: nullptr mosResource->bo
[LIBVA]:CRITICAL - MediaSurfaceToMosResource:66: nullptr mosResource->bo
0:00:00.066731031 393409 0x7f6ffb2db980 ERROR              vaencoder gstvaencoder.c:384:gst_va_encoder_open:<vaencoder1> vaCreateConfig: attribute not supported
0:00:00.066739506 393409 0x7f6ffb2db980 ERROR              vah265enc gstvah265enc.c:4573:gst_va_h265_enc_reconfig:<vah265lpenc0> Failed to open the VA encoder.
0:00:00.066742230 393409 0x7f6ffb2db980 ERROR              vabaseenc gstvabaseenc.c:611:gst_va_base_enc_reset:<vah265lpenc0> Error at reconfiguration error
0:00:00.066794991 393409 0x7f6ffb2db980 ERROR              vaencoder gstvaencoder.c:384:gst_va_encoder_open:<vaencoder1> vaCreateConfig: attribute not supported
0:00:00.066800987 393409 0x7f6ffb2db980 ERROR              vah265enc gstvah265enc.c:4573:gst_va_h265_enc_reconfig:<vah265lpenc0> Failed to open the VA encoder.
0:00:00.066802852 393409 0x7f6ffb2db980 ERROR              vabaseenc gstvabaseenc.c:611:gst_va_base_enc_reset:<vah265lpenc0> Error at reconfiguration error
0:00:00.066850798 393409 0x7f6ffb2db980 ERROR              vaencoder gstvaencoder.c:384:gst_va_encoder_open:<vaencoder1> vaCreateConfig: attribute not supported
0:00:00.066854457 393409 0x7f6ffb2db980 ERROR              vah265enc gstvah265enc.c:4573:gst_va_h265_enc_reconfig:<vah265lpenc0> Failed to open the VA encoder.
0:00:00.066857987 393409 0x7f6ffb2db980 ERROR              vabaseenc gstvabaseenc.c:611:gst_va_base_enc_reset:<vah265lpenc0> Error at reconfiguration error
ERROR: from element /GstPipeline:pipeline0/GstH265Parse:h265parse0: Internal data stream error.
Additional debug info:
../../subprojects/gstreamer/libs/gst/base/gstbaseparse.c(3694): gst_base_parse_loop (): /GstPipeline:pipeline0/GstH265Parse:h265parse0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
5. Running the same command with same stack on TGL-U are fine

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

1. libva/libva-utils - 2.16, gmmlib - 12.2.8107.12554, media-driver - commit 33b5939ca
2. ls /dev/dri
by-path  card0  renderD128
3. vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.16.0
libva info: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_16
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.6.4 (9042c1745)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP
4. [libva_trace.log.140007.zip](https://github.com/intel/media-driver/files/10163266/libva_trace.log.140007.zip)

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2022-12-06:
Auto Created VSMGWL-59721 for further analysis.

UngTengEn, 2023-01-04:
Hi, Is there any update?

UngTengEn, 2023-02-28:
This issue seems no longer reproduce with media driver 23.1.2 and on kernel 6.0.

Jexu, 2023-03-01:
Thanks for your inputs and could you please help close this issue if it works for you.

UngTengEn, 2023-03-01:
Ok, sure.

closed 2023-03-01 by UngTengEn