intel/media-driver #1572: [Bug][DG2][ffmpeg]: Blending BGRA on top of the NV12/BGRA produces wrong result

nyanmisaka, 2022-12-08:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. Linux only. (Occur in both Prod KMD and drm-tip kernel)
2. libva 2.16, media-driver 22.6.4 or git
3. Run ffmpeg (master/cartwheel) cli, VAAPI only, works fine in QSV.

Blend 0x00000000 Black with alpha value 0 on top of the Blue background. Alpha set to 1 or 255.

overlay_vaapi, BGRA on NV12, wrong
```
ffmpeg -init_hw_device vaapi=va -filter_hw_device va -f lavfi -i color=c=0x00000000:s=960x540,format=bgra -f lavfi -i color=c=blue:s=1920x1080,format=nv12 -filter_complex "[0:v]hwupload[ov];[1:v]hwupload[base];[base][ov]overlay_vaapi=alpha=1,hwdownload,format=nv12" -f image2 -vframes 1 -y va_bgra_on_nv12_wrong.png
```
[image]
---
overlay_vaapi, BGRA on BGRA, wrong
```
ffmpeg -init_hw_device vaapi=va -filter_hw_device va -f lavfi -i color=c=0x00000000:s=960x540,format=bgra -f lavfi -i color=c=blue:s=1920x1080,format=bgra -filter_complex "[0:v]hwupload[ov];[1:v]hwupload[base];[base][ov]overlay_vaapi=alpha=1,hwdownload,format=bgra" -f image2 -vframes 1 -y va_bgra_on_bgra_wrong.png
```
[image]
---
overlay_qsv, BGRA on NV12, ok
```
ffmpeg -init_hw_device qsv=qs -filter_hw_device qs -f lavfi -i color=c=0x00000000:s=960x540,format=bgra -f lavfi -i color=c=blue:s=1920x1080,format=nv12 -filter_complex "[0:v]hwupload=extra_hw_frames=16[ov];[1:v]hwupload=extra_hw_frames=16[base];[base][ov]overlay_qsv=alpha=255,hwdownload,format=nv12" -f image2 -vframes 1 -y qsv_bgra_on_nv12_ok.png
```
[image]
---
overlay_qsv, BGRA on BGRA, ok
```
ffmpeg -init_hw_device qsv=qs -filter_hw_device qs -f lavfi -i color=c=0x00000000:s=960x540,format=bgra -f lavfi -i color=c=blue:s=1920x1080,format=bgra -filter_complex "[0:v]hwupload=extra_hw_frames=16[ov];[1:v]hwupload=extra_hw_frames=16[base];[base][ov]overlay_qsv=alpha=255,hwdownload,format=bgra" -f image2 -vframes 1 -y qsv_bgra_on_bgra_ok.png
```
[image]

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

Wrong color when blending PGS subtitles into the videos.

### Debug Information

[vainfo.txt](https://github.com/intel/media-driver/files/10181115/vainfo.txt)
Everything else (dec, enc and other vpp features) works fine with DG2 on Linux in drm-tip.

[dmesg_i915.txt](https://github.com/intel/media-driver/files/10181123/dmesg_i915.txt)
No hang. Guc and Huc are enabled by default. Latest firmware from git.

Note that The same ffmpeg VAAPI cli works fine on older platforms like KBLx and TGLx. So it seems to be DG2 specific.

Also, not only BGRA and NV12, but also RGBA, ARGB and P010 are affected too.

### Do you want to contribute a patch to fix the issue?

None

FurongZhang, 2022-12-12:
@LhGu , please help to check this issue.

LhGu, 2022-12-23:
For overlay_qsv,  VA_BLEND_PREMULTIPLIED_ALPHA has been disabled in msdk:
[image]

There may be some hw issues in Per-pixel alpha case. 

For overlay_vaapi, it still use VA_BLEND_PREMULTIPLIED_ALPHA flag, you could change some codes in ffmepg: libavfilter\vf_overlay_vaapi.c:
[image]

nyanmisaka, 2022-12-23:
I will try disable `VA_BLEND_PREMULTIPLIED_ALPHA` in vf_overlay_vaapi and get back to you.

LhGu, 2022-12-23:
Why only Dg2 occurs this issue? For VA_BLEND_PREMULTIPLIED_ALPHA, I think DG2 could only use Dscale+Premultiplied kernel. AVS and common scale kernel have been removed in DG2.  May some hw limitations in Dscale+Premultiplied when our alpha value is zero.

nyanmisaka, 2022-12-23:
Disabling `VA_BLEND_PREMULTIPLIED_ALPHA` in overlay_vaapi does the trick.
Verified on DG2 and Kabylake.

LhGu, 2022-12-28:
From wrong images, I think the alpha value of overlay image also affects main image in DG2.  Change codes in ffmpeg to align with qsv.  If we disable VA_BLEND_PREMULTIPLIED_ALPHA flag, it will use BLEND_SOURCE and I also confirmed that it could work. I will close this issue.

closed 2022-12-28 by LhGu

FurongZhang, 2022-12-29:
Thank you so much @LhGu for checking and close the issue.