intel/media-driver #1575: [Bug]: ffempg-vaapi hevc 8bit/10bit encode fail on MTL

Wanghangjie8, 2022-12-14:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

ffempg-vaapi hevc 8bit/10bit encode fail on MTL

cmd line:
ffmpeg -v verbose -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format vaapi -f rawvideo -pix_fmt yuv420p -s:v 720x480  -r:v 30 -i input -vf 'format=nv12,hwupload' -an -c:v hevc_vaapi -rc_mode CBR -profile:v main -g 30 -slices 1 -b:v 3500k -maxrate 3500k -low_power 1 -vframes 150 -y output

error log:
Failed to create encode pipeline configuration: 10 (attribute not supported).
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height

The same command line has no problem on other platforms, such as ADL/TGL

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Wanghangjie8, 2022-12-14:
reproduce env:
Kernel | 5.15.47+prerelease4236 |  
libva | https://github.com/intel/libva.git | aeb704af762e3b5010908a544f5b2b1adeb27896
Libva-utils | https://github.com/intel/libva-utils.git | c0ed286c0c56bb9bcc787c88a87be9ad8571da9d
gmmlib | https://github.com/intel/gmmlib | ed21f2fd2e6c6e394fe5eb6b47d5ad4cb42668b5
media-driver | https://github.com/intel/media-driver | 7cef45fcc351cdecc7e7ce6ecba0e70f7afb4efa

Wanghangjie8, 2022-12-26:
Hello, everyone. Have you made any progress?

Wanghangjie8, 2023-01-10:
I tried to use the 5.15.47+prerelease4327 kernel, but the problem of ffmpeg-vaapi hevc 8bit/10bit encode still exists.

Wanghangjie8, 2023-01-13:
gst-va hevc 8bit/10bit encode fail on MTL , It is the same as the error message of ffmpeg-vaapi hevc 8bit/10bit encode.

Wanghangjie8, 2023-01-18:
gst-va cmd line: gst-launch-1.0 -vf filesrc location=input.yuv num-buffers=150 ! rawvideoparse format=i420 width=720 height=480 framerate=30 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=NV12 ! vah265lpenc rate-control=vbr key-int-max=30 target-usage=4 num-slices=3 bitrate=6000 target-percentage=70 ref-frames=1 ! video/x-h265,profile=main ! h265parse ! filesink location=output.h265

Wanghangjie8, 2023-01-18:
ffmpeg-vaapi cmd line:ffmpeg -v verbose -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format vaapi -f rawvideo -pix_fmt p010le -s:v 1920x1080  -r:v 30 -i input.P010 -vf 'format=p010le,hwupload' -an -c:v hevc_vaapi -rc_mode CBR -profile:v main10 -g 100 -slices 1 -b:v 3500k -maxrate 3500k -low_power 1 -vframes 150 -y output.h265

XinfengZhang, 2023-02-01:
fixed by https://github.com/intel/media-driver/commit/28379eaccac518e4a1b5d37c61fd40dda1751292

closed 2023-02-17 by Wanghangjie8