intel/media-driver #1576: [Bug]:

Deathproof76, 2022-12-14:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

Wenn trying to trancode a h264 or hevc file via FFmpeg the following error can be seen in the output:

`[h264_qsv @ 0x5633a4634580] A decode call did not consume any data: expect more data at input (-10)`

I'm using FFmpeg with cartwheel patches applied with the following configuration:
 
`configuration: --prefix=/home/max/.local/ --extra-cflags=-I/opt/intel/mediasdk/include --extra-ldflags=-L/opt/intel/mediasdk/lib --extra-ldflags=-L/opt/intel/mediasdk/plugins --enable-vaapi --enable-opencl --disable-debug --enable-libvpl --enable-libvorbis --enable-libvpx --enable-libdrm --enable-gpl --cpu=native --enable-libfdk-aac --enable-libx264 --enable-libx265 --extra-libs=-lpthread --enable-nonfree`

The resulting file works quality/filesize are as expected but has only a single I-Frame of the beginning of the file which leads to problems when seeking during playback.[ See here for further info.](https://www.reddit.com/r/ffmpeg/comments/zjwm4r/trouble_with_ffmpeg_using_quicksync_encoded_files/) Also here [an affected sample file ](https://send.vis.ee/download/06403a4b159a5ccd/#6BjKFjmUrJAiKaMdxe3A8w) (simpsons movie trailer).

Output of ffprobe -show_frames input.mkv -hide_banner | grep pict_type=I
Input #0, matroska,webm, from 'problem.mkv':
  Metadata:
    title           : The Simpsons Movie
    ARTIST          : Fox
    DATE_RELEASED   : 2007
    GENRE           : Trailer
    ENCODER         : Lavf59.34.101
  Duration: 00:02:17.28, start: 0.000000, bitrate: 2562 kb/s
  Stream #0:0: Video: hevc (Main), yuv420p(tv, bt709, progressive), 1920x800 [SAR 1:1 DAR 12:5], 23.98 fps, 23.98 tbr, 1k tbn (default)
    Metadata:
      ENCODER         : Lavc59.52.102 hevc_qsv
      DURATION        : 00:02:17.262000000
  Stream #0:1: Audio: aac (LC), 44100 Hz, stereo, fltp (default)
    Metadata:
      title           : Stereo
      DURATION        : 00:02:17.276000000
pict_type=I

Vaapi transcode seems to be working

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

Env: Ubuntu 22.04 with 5.17.0-1019-oem using the [gpu-backport.](https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-jammy-arc.html)
also current and built from source media-driver with -DENABLE_PRODUCTION_KMD=ON, gmmlib, libva, Libva utils

```
ls /dev/dri
by-path  card0  card1  renderD128  renderD129
```

```
vainfo --device /dev/dri/renderD129 --display drm
Trying display: drm
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.17.0.1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.6.4 (7cef45fcc)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointEncSlice
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointEncSlice
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointEncSlice
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP
```

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2022-12-19:
This is probably vpl/ ffmpeg issue which doesn't read any valid data from bitstream. As above reference, similar report has been in vpl repo.

Deathproof76, 2022-12-19:
Yes, that may be - the references are from me. But I'm not sure.

nyanmisaka, 2023-01-03:
> This is probably vpl/ ffmpeg issue which doesn't read any valid data from bitstream.

I came across this issue on DG2 these days.

It seems that the VPL runtime does not set a proper `mfx.GopPicSize` internally when [ffmpeg](https://github.com/FFmpeg/FFmpeg/blob/c94988a7814a8268a4732725472bed8b3aa1ba74/libavcodec/qsvenc_hevc.c#L376) `gop_size` is -1, which is the default value of all QSV encoders. As a consequence, it produces "I B B B B B B B B..." frames with only one I frame at the beginning.

nyanmisaka, 2023-01-03:
I can only find the `mfx.GopPicSize == 0` handling in H264 and MPEG2 in [VPL codes](https://github.com/oneapi-src/oneVPL-intel-gpu/tree/main/_studio/mfx_lib/encode_hw).
So the issue cannot be reproduced in these codecs.
Maybe you guys should check out `mfx.GopPicSize` for the rest of the codecs such as HEVC and AV1.

[H264 encode_hw](https://github.com/oneapi-src/oneVPL-intel-gpu/blob/01a2422f1aa716456012ef03b32da6aa5d5fc511/_studio/mfx_lib/encode_hw/h264/src/mfx_h264_enc_common_hw.cpp#L5662-L5668)
```
    if (par.mfx.GopPicSize == 0)
    {
        mfxU32 maxScale = (par.calcParam.numTemporalLayer) > 0
            ? par.calcParam.scale[par.calcParam.numTemporalLayer - 1]
            : 1;
        par.mfx.GopPicSize = mfxU16((256 + maxScale - 1) / maxScale * maxScale);
    }
```

[MPEG2 encode_hw](https://github.com/oneapi-src/oneVPL-intel-gpu/blob/01a2422f1aa716456012ef03b32da6aa5d5fc511/_studio/mfx_lib/ext/mpeg2/src/mfx_mpeg2_encode_utils_hw.cpp#L104-L107)
```
    if (par->mfxVideoParams.mfx.GopPicSize == 0)
    {
        par->mfxVideoParams.mfx.GopPicSize = 24;
    }
```

Jexu, 2023-01-04:
Need a fix in vpl for GPO size check.

OttoYang, 2023-03-13:
The default `GopPicSize` in HEVCe is (mfxU16)INFINITE(0xFFFF).  
You can find this in [HEVC hevcehw_base_legacy_defaults](https://github.com/oneapi-src/oneVPL-intel-gpu/blob/main/_studio/mfx_lib/encode_hw/hevc/agnostic/base/hevcehw_base_legacy_defaults.cpp#L180)

```c
    static mfxU16 GopPicSize(
        Defaults::TChain<mfxU16>::TExt
        , const Defaults::Param& par)
    {
        if (par.mvp.mfx.GopPicSize)
        {
            return par.mvp.mfx.GopPicSize;
        }
        const mfxExtHEVCParam* pHEVC = ExtBuffer::Get(par.mvp);
        if (par.mvp.mfx.CodecProfile == MFX_PROFILE_HEVC_MAINSP ||
            Legacy::IsMain10SP(par.mvp.mfx.CodecProfile, pHEVC))
        {
            return 1;
        }
        return mfxU16(GOP_INFINITE);
    }
```

So if you want to adjust it to suit your scenario, you can explicitly set `GopPicSize`.

nyanmisaka, 2023-03-13:
https://github.com/FFmpeg/FFmpeg/commit/a3c0a3ec604da635fba5d356d7ec53bbd7c38d78

But the ffmpeg-qsv changes want the runtime to set the suitable GOP size. Apparently 0xffff shouldn’t be used as default.

wenbinc-Bin, 2023-03-13:
I remember this change. We synchronized with VPL team and decoded to use the VPL runtime's default setting. Any improvement on VPL runtime can work on APP immediately, and it also helps to keep quality consistent on different APPs, such as ffmpeg and gstreamer.
I think we need to set a reasonable gop value in VPL runtime.

justin8, 2023-05-29:
I also see this on an Intel iGPU; I thought it was creating broken videos that crash/freeze apps on seeking but it was just trying to stream gigabytes of b-frames to catch up.

Even manually setting the I-frame interval to 2 seconds with `-force_key_frames expr:gte(t,n_forced*2)` results in a single I-frame when doing HEVC encoding.

rmeissn, 2023-06-24:
I think I'm also affected by this and I'm not using a DG2, but a regular 12th gen processor with integrated graphics.
Seeking on hevc_qsv encoded mkv is terribly slow and feels like the playback program froze. Seeking on libx265 encoded mkv is blazing fast.

rmeissn, 2023-06-26:
I got a i7-1260p with integrated ADL GT2 graphics (in a Framework 13, 2. gen)

justin8, 2023-06-26:
And I'm seeing it on an i5-11600K

Deathproof76, 2023-06-27:
For anyone stumbling upon this looking for a quick fix: put something like `"-g:v 120"` in your ffmpeg line. 

Like: `ffmpeg -hwaccel_output_format qsv -qsv_device /dev/dri/renderD128 -c:v h264_qsv -i input.mkv -map 0 -b:v 2500k -minrate 500k -maxrate 5000k -g:v 120 -c:v hevc_qsv -c:a copy -c:s copy output.mkv`

Deathproof76, 2023-06-27:
> And I'm seeing it on an i5-11600K

most likely because you're trying to use quick sync with a --enable-libvpl compiled ffmpeg

fawick, 2023-07-04:
For what it's worth: The setting `-g:v 120` does not help for codec `vp9_qsv`, any FFmpeg-created video encoded with `vp9_qsv` will not seek correctly and cause some video players to terminate prematurely with an error.

Qubelia, 2023-07-23:
Came stumped upon this post while debugging.
I'm encountering the same issue but on the infamous DG1 card(ARC is DG2), `hevc_qsv` transcoding(decode+encode) a simple mp4 clip with ffmpeg yields in a broken video with only the start of video working just as OP stated. My CPU is Celeron G3900 with iGPU disabled.
An additional weird issue is that after transcoding session had ended(GPU engine load returns 0), the CPU is pegged at high load by ffmpeg for some time. But all problem vanish with `-g:v 120`, though I'm on Windows 10, not Linux.
The transcode command I use is `-c:v hevc_qsv -preset medium -global_quality 22 -look_ahead 1 -color_range pc -g:v 120 -c:a copy`.
I used ffbatch to take care of input commands and selected `qsv` for decoding.(essentially the same as `-hwaccel_output_format qsv`) Interestingly enough handbrake doesn't exhibit the same problem, though CPU load is high with HB.

OttoYang, 2023-07-24:
@Deathproof76 @justin8 @eero-t @rmeissn @fawick @Qubelia 
Thanks for your information, the impact of the solution on quality and performance is being discussed and will hopefully be fixed soon.

MatthewRobertDunn, 2023-09-30:
latest ffmpeg still does this....makes qpv_vp9 basically unusable :(

warduweram, 2023-10-20:
I think it could be related to the transition from VA-API to Intel OneAPI, everything seems to be quite chaotic atm. Intel might not even want people to have broad access to the capability on budget CPUs. Maybe ffmpeg can still pull it off at some point, because this would be so valuable, it's very energy efficient and fast.

nyanmisaka, 2023-10-20:
https://github.com/FFmpeg/FFmpeg/commit/a7f24d79b8eeb4f4ab78916ef08cc997f2381cc9
This FFmpeg commit should have fixed this issue.
I think this can be closed now? cc @OttoYang

warduweram, 2023-10-20:
@nyanmisaka Oh, sorry, I wasn't aware I can't use qsv on a libvpl-compiled version of ffmpeg. This all goes a bit over my head, could you maybe point me to where I can get a most recent ffmpeg with libmfx without having to compile it myself? Can I even use it to properly hardware encode vp9_qsv under Windows with a 13th gen i5 and recent Intel graphics drivers? I use a GeForce as display GPU if that makes any difference. Thanks for the input in advance and sorry to spam the thread here with this. :D

nyanmisaka, 2023-10-20:
https://github.com/BtbN/FFmpeg-Builds/releases

warduweram, 2023-10-20:
@nyanmisaka And which one do I download? I tried 6.0 lgpl and N gpl and they both have libvpl enabled, not libmfx. Where do I see in the nomenclature that it's libmfx instead?

eero-t, 2023-10-20:
> @nyanmisaka And which one do I download? I tried 6.0 lgpl and N gpl and they both have libvpl enabled, not libmfx. Where do I see in the nomenclature that it's libmfx instead?

There are two components for MFX and oneVPL; the frontend API (dispatcher), and the backend.

OneVPL's libvpl frontend (mfxVersion 2.x) API is nowadays preferred over libmfx (mfxVersion 1.x) API for everything.

As to *backends*, oneVPL frontend uses legacy MFX backend for old integrated GPUs, and the new, actively maintained oneVPL backend for discrete GPUs, see: https://github.com/oneapi-src/oneVPL/blob/master/README.md

----

PS. MFX 2.x API in OneVPL is mostly same, it just drops some rarely used functionality from legacy MFX 1.x API [1], and adds device discovery support. Latter is important in multi-GPU setups as app could then automatically get device with right capabilities (app code needs to be changed to take advantage of that, so all do not support it yet).

[1] See: https://www.intel.com/content/www/us/en/docs/onevpl/upgrade-from-msdk/2023-1/transition-from-intel-r-media-sdk-to-intel-r.html

warduweram, 2023-10-20:
@eero-t Thanks so much for taking the time to explain this properly, I shall investigate. :D

warduweram, 2023-10-21:
Alrighty, so there was some misunderstanding/miscommunication on my side:

My encodes actually do work (with libvpl enabled), but there's still issues with it and I thought they might be remedied by switching to libmfx. Firstly, let me clarify that I have these issues with VP9, not with hevc which seems to be fine (except the speed which is a lot slower on both hevc and VP9 with hardware acceleration than if I use Handbrake; I get about 50 fps compared to more than 4x that with Handbrake, but this might be an issue with the ffmpeg parameters, I'l keep trying around with that).

What definitely is not a matter of parameters, though, as has been described by others above, are the issues with moving the playhead while playing back the video, this will still break the video stream. Sometimes I get artifacts with VP9, sometimes the playback just breaks until I restart the file (hevc seems fine now with the new addition from Intel, but I wouldn't be surprised if there was still a bug here and there, which time will show).

So, VP9 is still unusable and I would emphatically like to ask for proper support for it, since it's a great codec, especially now with the high power efficiency and speed on the newer Intel iGPUs and Arcs. It has a far wider support across different devices than AV1 is going to have in the coming years, also better than hevc. I would also argue that it's better for archiving purposes all around than hevc. I'd very much appreciate if Intel and/or the ffmpeg devs could have another look at this.

OttoYang, 2023-10-30:
https://github.com/FFmpeg/FFmpeg/commit/a7f24d79b8eeb4f4ab78916ef08cc997f2381cc9
I'll close this hevc related issue as this FFmpeg commit should have fixed it.  
If you encounter other problems with other codecs, please feel free to report new ones, thank you.

closed 2023-10-30 by OttoYang