intel/media-driver #1579: [Bug]: Wrong result for AV1 filmgrain decoding

ndufresne, 2022-12-18:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

When testing VA decoders on using fluster conformance tests, all filmgrain tests fails. It seems like VA driver adds some "noise" to the image, but it does not correlate with AOM reference implementation. The film grain algorithm seems to be fully described in the spec also (7.18.3. Film grain synthesis process). In comparision, we have been able to test NVidia using their NVDEC API, and all these tests are pixel perfect. We also been able to test 8bit decoding with Verisillicon VC9000D, and gaint, these test are pixel perfect there.

```
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
VAProfileAV1Profile0            :	VAEntrypointVLD
```

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

This means that image reproduction accuracy will be degraded, it will also affect transcoding use cases.

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2022-12-19:
Which platform are you testing?

intel-mediadev, 2022-12-19:
Auto Created VSMGWL-60087 for further analysis.

ndufresne, 2022-12-19:
Intel Gen12th 12900K processor.

Jexu, 2022-12-20:
In Intel GPU, the TigerLake  based systems use a simplified film grain generation algorithm, which satisfies the statistic similarity requirement but does not guarantee the bit match. This is allowed by the spec definition. But Xe based systems will change to use bit match algorithm.

You can check more details from spec section 7.2 which lists two options for vender decoder to apply noise.

ndufresne, 2022-12-20:
I looked at the correlation, its pretty low, but if this is by design, its your choice, we will mark these test as known broken on this HW. Can you add some clarify on how to filter the HW generation that are doing this and the one that will use the accurate version ?

Jexu, 2022-12-22:
All Gen12th are using simple algorithm, but MTL is using accurate version.

Jexu, 2023-01-04:
As it is expected result by design, so I will close this issue now and feel free to re-open it again if having any concerns.

closed 2023-01-04 by Jexu