intel/media-driver #1580: [Bug]: SwizzleSurfaceByHW calls nullptr pfnMediaMemoryTileConvert without 'kernel shaders' enabled

zlice, 2022-12-19:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

- using shotcut/mlt7(melt7) to splice videos
- segfault in iHD_drv_video.so

trace led to `SwizzleSurfaceByHW` which at [media_libva_util.cpp#L1147](https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva_util.cpp#L1147) calls `pfnMediaMemoryTileConvert` which is defaulted to a [nullptr](https://github.com/intel/media-driver/blob/b4262af83b7b3368f3cb61c4bd56a4beef13924a/media_driver/linux/common/ddi/ddi_media_context.h#L150) and never set if [MMC_SUPPORTED](https://github.com/intel/media-driver/blob/b67820f3d36b12121cb81282cb47afa50ada28c4/media_softlet/linux/common/ddi/media_libva_interface_next.cpp#L368) is false

I added a `ifdef` around the 1147 call and was able to export a video with vaapi fine. Not sure if that is the right fix or not. I also didn't look to see if anything else is potentially called but unset anywhere.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

I really don't know. Running ffmpeg seemed to work fine. Firefox vaapi seemed to work fine but I have gotten a few crashes (guessing this is related)

### Debug Information

```
#0  0x0000000000000000 in ?? ()
#1  0x00007fff7f56b9d9 in SwizzleSurfaceByHW (surface=0x7fff840cf820, isDeSwizzle=<optimized out>) at /builddir/intel-media-driver-22.6.4/media_driver/linux/common/ddi/media_libva_util.cpp:1146
#2  0x00007fff7f56c341 in DdiMediaUtil_LockSurfaceInternal (surface=surface@entry=0x7fff840cf820, flag=flag@entry=3) at /builddir/intel-media-driver-22.6.4/media_driver/linux/common/ddi/media_libva_util.cpp:1224
#3  0x00007fff7f56c7a1 in DdiMediaUtil_LockSurface (flag=3, surface=0x7fff840cf820) at /builddir/intel-media-driver-22.6.4/media_driver/linux/common/ddi/media_libva_util.cpp:1168
#4  DdiMediaUtil_LockBuffer (buf=buf@entry=0x7fff84203540, flag=flag@entry=3) at /builddir/intel-media-driver-22.6.4/media_driver/linux/common/ddi/media_libva_util.cpp:1384
#5  0x00007fff7f559e97 in DdiMedia_MapBufferInternal (ctx=<optimized out>, buf_id=<optimized out>, pbuf=0x7fff88b24540, flag=3) at /builddir/intel-media-driver-22.6.4/media_driver/linux/common/ddi/media_libva.cpp:3624
#6  0x00007fff8c2e30a6 in vaMapBuffer () from /usr/lib/libva.so.2
#7  0x00007fff8e2233dd in ?? () from /usr/lib/libavutil.so.56
#8  0x00007fff8e223b4a in ?? () from /usr/lib/libavutil.so.56
#9  0x00007fff8e218590 in av_hwframe_transfer_data () from /usr/lib/libavutil.so.56
#10 0x00007fff8cc09b96 in ?? () from /usr/lib/libavfilter.so.7
#11 0x00007fff8cb4128e in ?? () from /usr/lib/libavfilter.so.7
#12 0x00007fff8cb44e8d in ?? () from /usr/lib/libavfilter.so.7
#13 0x00007fff8e7bbaf5 in ?? () from /usr/lib64/mlt-7/libmltavformat.so
#14 0x00007ffff7c24afa in start_thread (arg=<optimized out>) at pthread_create.c:442
#15 0x00007ffff7ca573c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

zlice, 2022-12-19:
my fix may export corrupt videos. so that is not viable. (green video)

Sherry-Lin, 2023-03-22:
This issue should be fixed by [e68feab](https://github.com/intel/media-driver/commit/e68feab7d). Please let us know if issue still exist on master tip.

intel-mediadev, 2023-03-22:
Auto Created VSMGWL-62744 for further analysis.

zlice, 2023-03-22:
At first glance master branch seems to run fine. I'll let you know if I run into any issues. Thanks.

zlice, 2023-03-23:
I have used this today without any noticeable issues. :+1:

closed 2023-03-23 by Sherry-Lin