intel/media-driver #1581: [Bug]: 22.6.5-git DG2 decoding regression in c0f858c

nyanmisaka, 2022-12-19:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Use ffmpeg vaapi to decode any video on DG2.

DG2 decoding fails with the recently added commit c0f858c.

```
        uint64_t isRecoverableContextEnabled = 0;
        mos_get_context_param(osDriverContext->intel_context, 0, I915_CONTEXT_PARAM_RECOVERABLE, &isRecoverableContextEnabled);
        //// This return -22(EINVAL) since 'osDriverContext->intel_context' is nullptr

        // set recoverablecontext disabled if want disable object capture
        if (MEDIA_IS_WA(&m_waTable, WaDisableSetObjectCapture) && isRecoverableContextEnabled)
        {
            mos_bufmgr_gem_disable_object_capture(m_bufmgr);
            //// This must be applied on DG2 otherwise decoding fails.
            //// It used to be controlled by 'MEDIA_IS_SKU(&m_skuTable, FtrLocalMemory)'
        }
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2022-12-20:
What failure it is? Can you share more failure logs?

nyanmisaka, 2022-12-20:
> What failure it is? Can you share more failure logs?

Failed to end picture decode issue: 23 (internal decoding error)
Failed to end picture decode issue: 23 (internal decoding error)
Failed to end picture decode issue: 23 (internal decoding error)
...

Once I revert c0f858c the issue will disappear.

Jexu, 2022-12-20:
Problem is caused by uninitialized osDriverContext->intel_context, will add strict check there.

nyanmisaka, 2022-12-27:
Fixed by 6b210dd

closed 2022-12-27 by nyanmisaka