intel/media-driver #1583: [Bug]: encoding doesn't work on a A750 GPU, caused by alignment mismatch between kernel and media-driver

anszom, 2022-12-29:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

GPU: DG2 Arc A750 (8086:56a1)
Kernel: 6.1.0 from Linus' github.
Media-driver: git version 7cef45fcc351cdecc7e7ce6ecba0e70f7afb4efa
Other components come from Linux mint 21 distribution: libva 2.14, ffmpeg 4.4.2

For the purpose of reproducing the bug, I'm using a simple ffmpeg command:

`ffmpeg -i inputvideo.mp4 -init_hw_device vaapi=/dev/dri/renderD128 -vf 'format=nv12|vaapi,hwupload' -c:v h264_vaapi output.mp4 `

With a debug build, I see the following messages:

> 
> [MOS]:  CRITICAL - SubmitCommandBuffer:1447: Command buffer submission failed!
> [CODEC]:CRITICAL - SubmitCommandBuffer:980: MOS returned error, eStatus = 0x23
> [CODEC]:CRITICAL - ExecuteSliceLevel:6525: MOS returned error, eStatus = 0x23
> [CODEC]:CRITICAL - ExecuteEnc:5098: Slice level encoding failed.
> [CODEC]:CRITICAL - Execute:675: MOS returned error, eStatus = 0x23
> [LIBVA]:CRITICAL - EncodeInCodecHal:1168: DDI:Failed in Codechal!
> [LIBVA]:CRITICAL - EndPicture:81: DDI:DdiEncode_EncodeInCodecHal return failure.
> [h264_vaapi @ 0x55f01b08c480] Failed to end picture encode issue: 24 (internal encoding error).
> [h264_vaapi @ 0x55f01b08c480] Encode failed: -5.
>

strace shows that ioctl(I915_GEM_EXECBUFFER2_WR) returns EINVAL

I've traced the source of the error to an alignment check in [i915_vma_insert](https://github.com/torvalds/linux/blob/v6.1/drivers/gpu/drm/i915/i915_vma.c#L804). In my configuration, the check requires the offset to be 2M-aligned, but media-driver uses 64K alignment.

I've made quick workaround by forcing 2M alignment in [mos_bufmgr.c](https://github.com/intel/media-driver/blob/master/media_softlet/linux/common/os/i915_production/mos_bufmgr.c#L3764) (also [here](https://github.com/intel/media-driver/blob/master/media_softlet/linux/common/os/i915/mos_bufmgr.c#L3686) because I don't know which one is actually used) and it seems to work.

I have no idea if the "correct" alignment is 2MB or 64KB, this would indicate if this is a kernel or userspace bug.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

All encoding scenarios are affected, I haven't tried decoding.

### Do you want to contribute a patch to fix the issue?

None

nyanmisaka, 2022-12-30:
Linux 6.1 doesn't fully support Arc yet, you can try drm-tip or 6.2-rc1 instead. Also the linux firmware must be updated.

See also https://github.com/intel/media-driver/issues/1487#issuecomment-1341827125

anszom, 2022-12-30:
Thanks, drm-tip works good.

closed 2022-12-30 by anszom