intel/media-driver #1584: [Bug]: No apparent way to build release without debug info

spiderkeys, 2022-12-29:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

None

### What happened?

When configuring and building the 22.5.4 release with any of the following CMake invocations, debug info is always included in the final build:

```bash
cmake .. -DBUILD_TYPE=Release
cmake .. -DCMAKE_BUILD_TYPE=Release
cmake .. -DBUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release
```
Example of the final build flags for a particular source file:
```
-Wreorder -Wsign-promo -Wnon-virtual-dtor -Wno-invalid-offsetof -fvisibility-inlines-hidden -fno-use-cxa-atexit -frtti 
-fexceptions -fpermissive -fcheck-new -std=c++1y -std=c++11 -O3 -DNDEBUG -fPIC -Werror -Wall -Winit-self 
-Wpointer-arith -Wno-unused -Wno-unknown-pragmas -Wno-comments -Wno-sign-compare -Wno-attributes 
-Wno-narrowing -Wno-overflow -Wno-parentheses -Wno-delete-incomplete -Werror=address -Werror=format-security 
-Werror=non-virtual-dtor -Werror=return-type -finline-functions -funswitch-loops -fno-short-enums -Wa,--noexecstack 
-fno-strict-aliasing -fmessage-length=0 -fvisibility=hidden -fdata-sections -ffunction-sections -Wl,--gc-sections -m64 
-DLINUX=1 -DLINUX -DNO_RTTI -DNO_EXCEPTION_HANDLING -DINTEL_NOT_PUBLIC -g -fstack-protector -D_AMD64_ 
-D__CT__ -D__linux__ -fno-tree-pre -fPIC -Wl,--no-as-needed -O2 -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 
-finline-limit=100 -MD -MT media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen11/codec/hal/codechal_kernel_hme_g11.cpp.o 
-MF CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen11/codec/hal/codechal_kernel_hme_g11.cpp.o.d 
-o CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen11/codec/hal/codechal_kernel_hme_g11.cpp.o 
-c /opt/intel/workspace/media-driver/media_driver/agnostic/gen11/codec/hal/codechal_kernel_hme_g11.cpp
```
As can be seen above, the -DCMAKE_BUILD_TYPE=Release does initially set the `O3 -DNDEBUG` flags, but then later on, the `media_compile_flags_linux.cmake` variable `MEDIA_COMPILER_FLAGS_COMMON` globally adds the `-g` flag regardless of build type.

Ultimately this results in a very large shared library binary (~350Mib):
```bash
bloaty /opt/mr/vaapi/lib/dri/iHD_drv_video.so
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  67.4%   236Mi   0.0%       0    .debug_info
   9.8%  34.2Mi   0.0%       0    .debug_loc
   8.0%  28.1Mi  72.3%  28.1Mi    .rodata
   4.0%  14.1Mi   0.0%       0    .debug_line
   3.2%  11.3Mi   0.0%       0    .debug_str
   2.1%  7.40Mi   0.0%       0    .debug_ranges
   1.9%  6.83Mi  17.6%  6.83Mi    .text
   1.6%  5.46Mi   0.0%       0    .debug_abbrev
   0.5%  1.88Mi   4.8%  1.88Mi    .data
   0.5%  1.61Mi   0.0%       0    .strtab
   0.2%   858Ki   2.2%   858Ki    .rela.dyn
   0.2%   705Ki   0.0%       0    .symtab
   0.2%   682Ki   1.7%   682Ki    .eh_frame
   0.2%   542Ki   0.0%       0    .debug_aranges
   0.1%   298Ki   0.7%   298Ki    .data.rel.ro
   0.0%   156Ki   0.4%   156Ki    .eh_frame_hdr
   0.0%  48.1Ki   0.1%  48.1Ki    .gcc_except_table
   0.0%  40.0Ki   0.1%  31.1Ki    [21 Others]
   0.0%  18.9Ki   0.0%  18.9Ki    .dynstr
   0.0%       0   0.0%  9.88Ki    .bss
   0.0%  8.88Ki   0.0%  8.88Ki    .dynsym
 100.0%   350Mi 100.0%  38.9Mi    TOTAL

```
whereas the version in the Ubuntu 20.04 distribution is only ~7MiB:
```bash
bloaty /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so 
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  48.5%  3.41Mi  48.5%  3.41Mi    .text
  37.9%  2.67Mi  37.9%  2.67Mi    .rodata
   4.9%   349Ki   4.9%   349Ki    .eh_frame
   4.5%   327Ki   4.5%   327Ki    .rela.dyn
   1.3%  92.7Ki   1.3%  92.7Ki    .data.rel.ro
   1.1%  76.9Ki   1.1%  76.9Ki    .eh_frame_hdr
   0.9%  66.9Ki   0.9%  66.9Ki    .data
   0.3%  18.1Ki   0.3%  18.1Ki    .gcc_except_table
   0.2%  11.2Ki   0.2%  11.2Ki    .dynstr
   0.0%       0   0.1%  9.74Ki    .bss
   0.1%  6.47Ki   0.1%  6.47Ki    .dynsym
   0.1%  6.01Ki   0.0%       0    [Unmapped]
   0.1%  3.98Ki   0.1%  3.98Ki    .rela.plt
   0.0%  2.68Ki   0.0%  2.35Ki    [12 Others]
   0.0%  2.67Ki   0.0%  2.67Ki    .plt
   0.0%  2.66Ki   0.0%  2.66Ki    .plt.sec
   0.0%  2.19Ki   0.0%  2.19Ki    .init_array
   0.0%  2.11Ki   0.0%  2.11Ki    .fini_array
   0.0%  1.81Ki   0.0%       0    [ELF Section Headers]
   0.0%  1.52Ki   0.0%  1.52Ki    .got
   0.0%     688   0.0%     688    .gnu.hash
 100.0%  7.03Mi 100.0%  7.03Mi    TOTAL
```
After manually deleting the `-g` flag, I was able to get the size down to ~41MiB
```bash
bloaty /opt/mr/vaapi/lib/dri/iHD_drv_video.so
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  68.5%  28.1Mi  72.5%  28.1Mi    .rodata
  16.3%  6.69Mi  17.2%  6.69Mi    .text
   4.6%  1.88Mi   4.8%  1.88Mi    .data
   3.9%  1.61Mi   0.0%       0    .strtab
   2.0%   858Ki   2.2%   858Ki    .rela.dyn
   1.7%   710Ki   1.8%   710Ki    .eh_frame
   1.7%   704Ki   0.0%       0    .symtab
   0.7%   298Ki   0.8%   298Ki    .data.rel.ro
   0.4%   156Ki   0.4%   156Ki    .eh_frame_hdr
   0.1%  48.0Ki   0.1%  48.0Ki    .gcc_except_table
   0.0%  18.9Ki   0.0%  18.9Ki    .dynstr
   0.0%       0   0.0%  9.88Ki    .bss
   0.0%  8.88Ki   0.0%  8.88Ki    .dynsym
   0.0%  7.41Ki   0.0%  7.41Ki    .rela.plt
   0.0%  6.09Ki   0.0%  3.82Ki    [14 Others]
   0.0%  4.95Ki   0.0%  4.95Ki    .plt
   0.0%  4.94Ki   0.0%  4.94Ki    .plt.sec
   0.0%  4.51Ki   0.0%       0    [Unmapped]
   0.0%  3.66Ki   0.0%  3.66Ki    .init_array
   0.0%  3.56Ki   0.0%  3.56Ki    .fini_array
   0.0%  2.73Ki   0.0%  2.73Ki    .got
 100.0%  41.1Mi 100.0%  38.8Mi    TOTAL
```
I do expect that this latest version will be larger than what is in the distribution, so the above numbers might be what one would expect.

Could someone comment on whether or not I'm missing some aspect around building a Release build without debug info, such that I don't need to manually modify the flags to get the expected binary? Otherwise, might it make sense to move the `-g` flag to the `MEDIA_COMPILER_FLAGS_RELEASEINTERNAL` or `MEDIA_COMPILER_FLAGS_DEBUG` variables?

Thanks

### What's the usage scenario when you are seeing the problem?

Others

### Debug Information

Git tags for dependencies:
libva = 2.16.0
gmmlib = intel-gmmlib-22.2.1

### Do you want to contribute a patch to fix the issue?

None

XinfengZhang, 2023-01-03:
how about use "strip" command to strip all of the debug symbols after compilation?

dvrogozh, 2023-01-03:
> how about use "strip" command to strip all of the debug symbols after compilation?

This is not an option. `-g` is a clear debug flag and should be removed from release configuration. Considering that this is managed directly by cmake, this should simply be dropped and added only to non-standard RELEASEINTERNAL configuration.

intel-mediadev, 2023-01-04:
Auto Created VSMGWL-60398 for further analysis.

eero-t, 2023-01-09:
> whereas the version in the Ubuntu 20.04 distribution is only ~7MiB:

All major distributions do release builds builds **with** debug symbols.   Debug symbols are split from binaries to separate debug package(s) during package builds, using distro's packaging tools.

If you build packages instead of just binaries, and install those packages, you should be getting stripped binaries.

spiderkeys, 2023-01-09:
I agree that stripping is what distributions generally do, and can be done in this case, but that is usually satisfied by the conventional CMake RelWithDebInfo configuration. 

As I mentioned, it is trivial to work around this, whether by modifying the CMake project or stripping binaries, but it just seems like the CMake Release configuration should be free of debug symbols, as is fairly conventional for most CMake projects.