intel/media-driver #1587: Libva & gmmlib requirements

spiderkeys, 2023-01-05:
Are there any particular version requirements for the combination of libva + gmmlib + media-driver?

I am building an application for both Ubuntu 20.04 and 22.04 and due to some constraints imposed by other system libraries, I need to use the distro versions of libva-dev, which are 2.7.0-2 and 2.14.0-1, respectively.

I would like to build gmmlib==21.3.5 and media-driver==21.4.2.

Would you expect any issues with this permutation of builds, or do I need to build these against libva==2.16?

eero-t, 2023-01-09:
(Note: I'm not media-driver developer)

Components may require some specific API from libva headers.    In general I would expect newer than expected libva version to work better than older, except for one API breakage issue last year.

AFAIK components are tested only against the recommended versions, so I think you just need to try and see how well they works.   If there are issues with libva, most likely you should see it already in the build phase.

PS. What problem you have in building libva on either 20.04 or 22.04: https://github.com/intel/libva/  ?

eero-t, 2023-03-02:
As the question is answered, I guess this can be closed?

closed 2023-03-23 by XinfengZhang