intel/media-driver #1592: [Feature]: Provide -av1qsv_params style AVOption to av1_qsv in ffmpeg to allow setting HDR10 mastering display metadata

millercentral, 2023-01-10:
### What Feature?

Today the av1_qsv encoder supports encoding 10-bit HDR video in the AV1 Main profile, however there does not appear to be any way to provide HDR10 mastering display metadata to the encoded content. 

With svt-av1, you can pass in the parameters in ffmpeg using the "-svtav1-params" command-line option, for example:
-c:v libsvtav1 -svtav1-params "preset=6:crf=13:tune=0:enable-hdr=1:color-primaries=9:transfer-characteristics=16:matrix-coefficients=9:mastering-display=G(0.265,0.69)B(0.15,0.06)R(0.68,0.32)WP(0.3127,0.329)L(4000.0,0.005):content-light=1000,755"
(svtav1 parameters are all documented here: https://gitlab.com/AOMediaCodec/SVT-AV1/-/blob/master/Docs/Parameters.md ). The x265 encoder has an equivalent option (x265-params) as well.

Please consider enabling an equivalent way to set mastering-display metadata on video created by AV1_QSV encoder

### What's the usage scenario would be benifited?

Transcode for media delivery, Content Creation, Game Streaming

### What impacted?

ffmpeg-based encoding techniques dealing with HDR10 video content would benefit from this.

### Do you want to contribute a patch to develop this feature?

None

XinfengZhang, 2023-05-06:
it is a feature request, from media driver perspective, the header is packed in app layer, and inserted into bitstream. 
so, suppose, this issue should be moved to oneVPL RT.

intel-mediadev, 2023-05-06:
Auto Created VSMGWL-64234 for further analysis.

leyu-yao, 2023-07-20:
OneVPL RT has added HDR support for av1 encoding. Implementation can be found at https://github.com/oneapi-src/oneVPL-intel-gpu/blob/main/_studio/mfx_lib/encode_hw/av1/agnostic/base/av1ehw_base_hdr.cpp.

millercentral, 2023-07-20:
Fantastic news, thank you.

leyu-yao, 2023-12-20:
Closing this issue because the requested feature is already implemented in VPL RT.

closed 2023-12-20 by leyu-yao