intel/media-driver #1593: [Bug]: Recreating an instance of OneVPL in a loop will eventually fail in Mos_Specific_RegisterResource

MathieuGirard, 2023-01-12:
### Which component impacted?

Decode, Encode, Video Processing

### Is it regression? Good in old configuration?

None

### What happened?

1. In linux
2. using media-driver 22.3.1
3. Sequence is creating a one VPL instance start an encoding or decoding process a few frame and destroy the one VPL instance, then restart that process (reusing the same VADisplay).

What expected is the long resilience test to run for a very long amount of time.
What we have is a failure in Mos_Specific_registerResource quite quickly since both HalCm_AllocateTsResource and HalCm_AllocateTrackerResource are registering their allocation and the attached ressource index is not reset (not even when we initialize the os interface).

Here is an example of the backtrace for each registered ressource the count is the number of times the stack trace was the same, the backtrace show the culprit execution path (in that example the test is decoding and encoding in H265 at each test execution).

The HAL interface is recreated at each test by media driver, but the OS specific attached surface database is never resetted.

n.b. A work around tested was to add a Mos_specific_ResetOsStates(pOsInterface)  at the end of Mos_initInterface. Which allows to run for much longer.

15 times: [
    "7ffff317a271 <HalCm_LockBuffer(_CM_HAL_STATE*, _CM_HAL_BUFFER_PARAM*)+302>",
    "7ffff31404d6 <CMRT_UMD::CmBuffer_RT::InitSurface(unsigned int, CMRT_UMD::CmEvent*)+742>",
    "7ffff3c6f203 <CodecHalHevcMbencG12::AllocateMeResources()+793>",
    "7ffff3c6e715 <CodecHalHevcMbencG12::AllocateEncResources()+341>",
    "7ffff3446263 <CodechalEncodeHevcBase::AllocateResources()+1177>",
    "7ffff336996b <CodechalEncoderState::Allocate(CodechalSetting*)+1843>",
    "7ffff3d2dcbb <CodechalEncHevcStateG12::Allocate(CodechalSetting*)+539>"]

94 times: [
    "7ffff315c6ed <HalCm_AllocateTsResource(_CM_HAL_STATE*)+397>",
    "7ffff317567c <HalCm_Allocate(_CM_HAL_STATE*)+1200>",
    "7ffff3e7cdfc <CMRT_UMD::CmDeviceRT::CreateAuxDevice(_MOS_OS_CONTEXT*)+254>",
    "7ffff3e7c56b <CMRT_UMD::CmDeviceRT::InitializeOSSpecific(_MOS_OS_CONTEXT*)+39>",
    "7ffff31446a2 <CMRT_UMD::CmDeviceRTBase::Initialize(_MOS_OS_CONTEXT*)+86>",
    "7ffff3e7c2a2 <CMRT_UMD::CmDeviceRT::Create(_MOS_OS_CONTEXT*, CMRT_UMD::CmDeviceRT*&, unsigned int)+174>",
    "7ffff3e87b02 <CreateCmDevice(_MOS_OS_CONTEXT*, CMRT_UMD::CmDevice*&, unsigned int)+82>"]

104 times: [
    "7ffff353ac24 <FrameTrackerProducer::Initialize(_MOS_INTERFACE*)+726>",
    "7ffff315ca4d <HalCm_AllocateTrackerResource(_CM_HAL_STATE*)+103>",
    "7ffff31756d6 <HalCm_Allocate(_CM_HAL_STATE*)+1290>",
    "7ffff3e7cdfc <CMRT_UMD::CmDeviceRT::CreateAuxDevice(_MOS_OS_CONTEXT*)+254>",
    "7ffff3e7c56b <CMRT_UMD::CmDeviceRT::InitializeOSSpecific(_MOS_OS_CONTEXT*)+39>",
    "7ffff31446a2 <CMRT_UMD::CmDeviceRTBase::Initialize(_MOS_OS_CONTEXT*)+86>",
    "7ffff3e7c2a2 <CMRT_UMD::CmDeviceRT::Create(_MOS_OS_CONTEXT*, CMRT_UMD::CmDeviceRT*&, unsigned int)+174>"]
 

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

After that point OneVPL and media driver becomes unstable unusable.

### Debug Information

List of version used:
oneVPL GPU Runtime: https://github.com/oneapi-src/oneVPL-intel-gpu/releases/tag/intel-onevpl-22.3.2
oneVPL Dispatcher and Samples: https://github.com/oneapi-src/oneVPL/releases/tag/v2022.0.3
Driver: https://github.com/intel/media-driver/releases/tag/intel-media-22.3.1
Gmmlib: https://github.com/intel/gmmlib/releases/tag/intel-gmmlib-22.1.2
libva: https://github.com/intel/libva/releases/tag/2.14.0
libva-utils: https://github.com/intel/libva-utils/releases/tag/2.14.0

ref: https://github.com/oneapi-src/oneVPL-intel-gpu/releases/tag/intel-onevpl-22.3.2

linux kernel: 5.18.0-051800-generic
Ubuntu 22.04 LTS

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2023-01-13:
Auto Created VSMGWL-60673 for further analysis.

MicroYY, 2023-01-13:
> Sequence is creating a one VPL instance start an encoding or decoding process a few frame and destroy the one VPL instance, then restart that process (reusing the same VADisplay).

What's your calling sequence of oneVPL API? Could you pls provide test app or sample code of the loop? And on which API the media driver failed? Did you call SyncOperation before destroying the session?

MathieuGirard, 2023-01-13:
The sample I have is running with a library we are developing here at
Matrox.

Yes we are waiting for all the synchronisation points.

Is synchronisation removing itself from the attached surface list? I have
not seen anything like that in the source code.

Regards,
Mathieu

MathieuGirard, 2023-01-13:
See line
https://github.com/intel/media-driver/blob/6547f46584633a7619440b39b34908e13c8c857d/media_driver/linux/common/os/mos_os_specific.c#L3625

MicroYY, 2023-03-06:
@MathieuGirard  May I know which platform you are using?

MathieuGirard, 2023-03-06:
It was on Linux.

MicroYY, 2023-03-07:
> 

I mean what's your hardware platform? Gen12 or other platforms？You can get it with:
`cat /proc/cpuinfo`
BTW， could you capture log for more debug info? **ReleaseInternal** build is needed.
`vim /etc/igfx_user_feature_next.txt `
and add following keys in it:
```
[config]
Message Print Enabled=1
Message HLT Enabled=1
Message HLT Output Directory=/data/vp_dump/
DDI Message Tags=7
Codec Message Tags=1
Mhw Message Tags=1
VP Message Tags=7
```
The `Message HLT Output Directory` can be adjusted to any path you prefer.
Then try to reproduce the issue without any extra driver changes (latest driver release is recommended).
Driver logs will be generated in the path you set.

MicroYY, 2023-05-18:
Since we cannot reproduce it and no logs are provided, we will close this issue.
@MathieuGirard If you want to fix this issue, you can reopen it and provide more details.

closed 2023-05-18 by MicroYY