intel/media-driver #1598: [Feature]: Consider partitioning encumbered codec into a separate library

Conan-Kudo, 2023-01-14:
### What Feature?

The Intel media driver compiles in all codecs into a single library, which makes it difficult for dealing with layered installations (`-free`/`-full` split) for AVC (H.264) and HEVC (H.265) support.

I'd like to ask if the driver library could be changed to check for an extra library somewhere to load that for AVC and HEVC support.

### What's the usage scenario would be benifited?

Playback, Web browser, Cloud Gaming, Video Conference, Content Creation, Game Streaming

### What impacted?

Fedora is working on introducing the `intel-media-driver-free` package and a third party wants to provide an overlay package to layer in the extra codecs. Swapping packages provides a poorer user experience (as we've discovered with `mesa-va-drivers` and `mesa-va-drivers-freeworld`) and we plan to preload the driver in Fedora Linux desktop variants to enable hardware acceleration of free codecs. So being able to install something that the main driver can opportunistically detect to load the encumbered codecs would make this much better.

### Do you want to contribute a patch to develop this feature?

None

kwizart, 2023-01-23:
Which ABI compatibility can be expected with this separate "encumbered library" ?

If none (strict matching), then I don't think it's an improvements as it will enforce a breakage induced by the update delay from two different provider. As we cannot provide as both updates will need to be provided at the same time.

Generally I don't see much benefit for the split only a separate problematic sub-content.

Conan-Kudo, 2023-01-23:
If it is dlopened, the ABI is necessarily not strict.

kwizart, 2023-01-23:
I would say the exact opposite as the "symbols" that will be part of this binary interface will need to be carefully defined.

Also it will break the common assumption that patches landing in this repository will be provided together.