intel/media-driver #1601: [Bug]: File handles can be leaked from calls to DdiMedia_ExportSurfaceHandle

Dave-Craig-Brightsign, 2023-01-27:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

When looking at file handle leaks from within MPV, I noticed that DdiMedia_ExportSurfaceHandle could leak handles too.

mos_bo_gem_export_to_prime is called which allocates file handles, but the return VADRMPRIMESurfaceDescriptor isn't filled with those file handles if the subsequent call to DdiMedia_MediaFormatToOsFormat fails. In that case VA_STATUS_ERROR_UNSUPPORTED_RT_FORMAT is returned with desc->num_objects set to 0.

I don't know in what situations this can happen, but it looks likely it's more than just a theoretical leak. This code is still present in commit aad03e1c19ccb56d01efe9ca154e8ee77b402273 which is the latest commit on libva.cpp.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

File handle leaking can be very severe, though as I don't know how to provoke the error handling code path I don't know how common this will be.

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

Jexu, 2023-01-30:
Yeah, it is good to release all allocated handles if some other calls fail that causing function exiting

intel-mediadev, 2023-01-30:
Auto Created VSMGWL-60909 for further analysis.

Jexu, 2023-02-16:
Normally descriptor is mananged by uplayter caller, but filled with details by umd. So is it possible that mpv helps release this file handle if DdiMedia_ExportSurfaceHandle call exists with failure. In this case, umd can fill descriptor with handle only.

closed 2023-03-08 by intel-mediadev

Dave-Craig-Brightsign, 2023-03-08:
This change is more dangerous than what was there before. To deal with the previous behaviour, applications have had to close any `fd` passed back, even if `DdiMedia_ExportSurfaceHandle` returned an error. With this new change `DdiMedia_ExportSurfaceHandle` is still returning the `fd`, but has already closed it. This means that the applications will now be calling `close(fd)` and either

1. The `fd` was already closed, or
2. If they are unlucky the `fd` has been reused by a different file and they close that
 
To fix this, when calling `close(fd)` the returned `num_objects` should be set to zero so that the application knows it doesn't have to do anything.

Jexu, 2023-03-09:
Since uncontroled behavior in applications, here umd takes the role to close the fd when unexpected error happens in umd. In this case, applications will receive error return as well, so we suggest applications don't try to close the fd anymore. Applications close maybe only required when umd returns success.

Dave-Craig-Brightsign, 2023-03-09:
> we suggest applications don't try to close the fd anymore
But if you just set `num_objects` to zero when there's an error no applications will try and close the `fd` and the code would work with old and new applications.

PR #1622 (2023-03-08): [Decode] Close the hanle if return error to avoid file handles leak
Fixes: #1601