intel/media-driver #1602: [Bug]: media-driver on kernel 6.2-rc6 and DG2 hardware fails execbuffer ioctl

airlied, 2023-02-02:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

With kernel 6.2.0-rc6 and latest media-driver on a DG2 system (tested on a380 and a770), av1 decode fails, digging in I found execbuffer was failing due to

i915_gem_execbuffer.c:eb_capture_stage is where it fails
```
                if (i915_gem_context_is_recoverable(eb->gem_context) &&
                    (IS_DGFX(eb->i915) || GRAPHICS_VER_FULL(eb->i915) > IP_VER(12, 0)))
                        return -EINVAL;
```
I assume this is media-driver doing something it shouldn't.

### What's the usage scenario when you are seeing the problem?

Playback

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2023-02-02:
This should be incompatible issue of i915 error capture and recoverable ctx involved by i915 change. Suppose media driver has disabled i915 error capure temporarily.
@Dylan-debug can you comment and share the commit.

nyanmisaka, 2023-02-02:
This was fixed in the latest tag is [intel-media-23.1.0](https://github.com/intel/media-driver/releases/tag/intel-media-23.1.0) https://github.com/intel/media-driver/commit/6b210dd1ec8b568f8760c80d31d5cf3ac86c501a.

22.6.5 and 22.6.6 are two broken versions for using DG2 with upstream kernel.

intel-mediadev, 2023-02-06:
Auto Created VSMGWL-61132 for further analysis.

airlied, 2023-02-06:
yes indeed updating to master seems to have fixed it, not sure why I dropped back to an earlier release, I was debugging something else.

closed 2023-02-06 by airlied