intel/media-driver #1608: [Bug]: vaInitialize fails on i5-1240P

chao-camect, 2023-02-04:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

On i5-1240P with kernel 6.0.18, vaInitialize returns "operation failed".
I tracked it down to `https://github.com/intel/media-driver/blob/intel-media-22.6.6/media_softlet/linux/common/os/hwinfo_linux.cpp#L235`

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

chao-camect, 2023-02-05:
I found following messages from dmesg, which might be relevant. Does this mean the firmware is missing in the kernel source code?
```
[    3.012338] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    3.012848] Console: switching to colour dummy device 80x25
[    3.012891] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.012927] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    3.013657] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.013704] i915 0000:00:02.0: Direct firmware load for i915/adlp_dmc_ver2_16.bin failed with error -2
[    3.013707] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/adlp_dmc_ver2_16.bin. Disabling runtime power management.
[    3.013709] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[    3.020673] i915 0000:00:02.0: GuC firmware i915/adlp_guc_70.1.1.bin: fetch failed with error -2
[    3.020676] i915 0000:00:02.0: Please file a bug on drm/i915; see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
[    3.020678] i915 0000:00:02.0: [drm] GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[    3.108644] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.1.1.bin version 0.0
[    3.108660] i915 0000:00:02.0: GuC initialization failed -2
[    3.108661] i915 0000:00:02.0: Enabling uc failed (-5)
[    3.108663] i915 0000:00:02.0: Failed to initialize GPU, declaring it wedged!
[    3.108836] i915 0000:00:02.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_gt_set_wedged_on_init+0x34/0x50 [i915]

```

dvrogozh, 2023-02-05:
Firmware is not stored in kernel source code. Kernel sources just have a firmware version which needs to be loaded. Firmware itself is stored on a file system in `/lib/firmware`. For i915 FW look in subfolder: `/lib/firmware/i915/`. Sometimes update mechanism stores FW in `/lib/firmware/updates/i915/`. So, to check whether FW named in dmesg is missing, run this:
```
find /lib/firmware -name adlp_dmc_ver2_16.bin
find /lib/firmware -name adlp_guc_70.1.1.bin
```
If it is missing (likely for your case), you have 2 ways to solve that:
1. Your OS should have FW package installed. Try to update it and see maybe required FW version will get pulled in. For ubuntu 22.04 that's `linux-firmware` package and it holds both FW files named above (see, https://packages.ubuntu.com/jammy/linux-firmware, look into list of files).
2. In case your OS does not yet provide FW package with required FW, you can pull it in manually following the process:
```
git clone https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
# now copy FW. Either copy everything or only those FW which is missing:
cp -rd linux-firmware/i915/* /lib/frimware/i915/
```

Now important part. Once you will fetch required firmware you need to regenerate initramfs otherwise kernel will still miss new FW you just copied. If you will fetch what you need updating OS package - there is a chance initramfs will get triggered for you - look into package manager log. Otherwise or if you copied firmware manually, you can regenerate initramfs as follows (for the particular kernel version you are interested in):
```
sudo update-initramfs -c -k $(uname -r)
```
Look into output of this command - it typically contain warnings for the FW files you still miss.

chao-camect, 2023-02-05:
Thanks for the quick response! One more question: is reboot required for this to take effect?

dvrogozh, 2023-02-05:
> One more question: is reboot required for this to take effect?

Yes, reboot is required.

chao-camect, 2023-02-06:
Thanks. It works.

closed 2023-02-06 by chao-camect