intel/media-driver #1610: [Bug]: Bitrate output in Windows is greater than that in Linux

wenqingx, 2023-02-13:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Windows and Linux use the same command line, but the bitrate output by Windows is larger than that of Linux.
The result of linux is normal.

cmdline
windows:ffmpeg.exe -v verbose -hwaccel qsv -init_hw_device qsv=qsv,child_device=0 -hwaccel_output_format qsv -f rawvideo -pix_fmt p010le -s:v 720x480  -r:v 25 -i input.P010 -vf 'format=p010le,hwupload=extra_hw_frames=120' -an -c:v hevc_qsv  -profile:v main10 -g 1 -bf 0 -slices 1 -b:v 500k -maxrate 500k -low_power 0 -vframes 100 -y ouput.h265
log:
DEBUG: slash:   Input stream #0:0 (video): 100 packets read (103680000 bytes); 100 frames decoded; 
DEBUG: slash:   Total: 100 packets (103680000 bytes) demuxed
DEBUG: slash: Output file #0 
DEBUG: slash:   Output stream #0:0 (video): 100 frames encoded; 100 packets muxed (285163 bytes); 
DEBUG: slash:   Total: 100 packets (**285163 bytes**) muxed

linux:ffmpeg -v verbose -hwaccel qsv -init_hw_device qsv=qsv,child_device=/dev/dri/renderD128 -hwaccel_output_format qsv -f rawvideo -pix_fmt p010le -s:v 720x480  -r:v 25 -i input.P010 -vf 'format=p010le,hwupload=extra_hw_frames=120' -an -c:v hevc_qsv  -profile:v main10 -g 1 -bf 0 -slices 1 -b:v 500k -maxrate 500k -low_power 0 -vframes 100 -y out.h265
log:
DEBUG: slash:   Input stream #0:0 (video): 100 packets read (103680000 bytes); 100 frames decoded; 
DEBUG: slash:   Total: 100 packets (103680000 bytes) demuxed
DEBUG: slash: Output file #0 
DEBUG: slash:  Output stream #0:0 (video): 100 frames encoded; 100 packets muxed (274160 bytes); 
DEBUG: slash:   Total: 100 packets (**274160 bytes**) muxed

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

XinfengZhang, 2023-02-13:
it is windows issue? not a open source media driver issue, right?

wenqingx, 2023-02-13:
> it is windows issue? not a open source media driver issue, right?

I'm not sure. Now it seems that Windows and Linux are inconsistent.
Hi @tong1wu ,Do you have any better suggestion?

wenqingx, 2023-02-13:
I will report this problem internally, closed. thanks

closed 2023-02-13 by wenqingx