intel/media-driver #1614: [Bug]: ffempg-vaapi AV1 8bit encode fail on MTL

Wanghangjie8, 2023-02-17:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

ffempg-vaapi av1 8bit encode fail on MTL

cmd line:
ffmpeg -v verbose -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format vaapi -f rawvideo -pix_fmt yuv420p -s:v 1920x1080  -r:v 30 -i 1080p_blue_sky_1920x1080_I420.yuv -vf 'format=nv12,hwupload' -an -c:v av1_vaapi -rc_mode CBR -profile:v main -compression_level 4 -g 30 -bf 0 -b:v 8000k -maxrate 8000k -tile_cols_log2 1 -tile_rows_log2 2 -low_power 1 -vframes 100 -y output.ivf

error log:
[av1_metadata @ 0x558a087f47c0] Invalid value at cdef_uv_pri_strength[i]: bitstream ended.d=N/A
[av1_metadata @ 0x558a087f47c0] Failed to read unit 1 (type 3).
[av1_metadata @ 0x558a087f47c0] Failed to read temporal unit from packet.
[ivf @ 0x558a082e4840] Error applying bitstream filters to an output packet for stream #0: Invalid data found when processing input
av_interleaved_write_frame(): Invalid data found when processing input
Error muxing a packet for output file #0

The same command line has no problem on the DG2 platform

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Wanghangjie8, 2023-02-17:
Reappearance env:
Kernel: 5.15.47+prerelease4327
libva: aeb704af762e3b5010908a544f5b2b1adeb27896
libva-utils: c0ed286c0c56bb9bcc787c88a87be9ad8571da9d
gmmlib: ed21f2fd2e6c6e394fe5eb6b47d5ad4cb42668b5
media-driver: b49f4eb07312f73ea3fae9b0f76e2ffd57bf0a44

feiwan1, 2023-02-17:
Noticed VAConfigAttribValEncAV1Ext2.obu_size_bytes_minus1 is queried as 0. Assume driver missed report some av1 caps for this platform.

Wanghangjie8, 2023-02-19:
Hi. Have you made any progress?

XinfengZhang, 2023-05-06:
suppose it already been fixed by d2ad171b86151bdee74c085cf2ef2a3ef725068e

closed 2023-05-06 by XinfengZhang