intel/media-driver #1615: [Bug]: i7-11700 decode 422 8bit problem

okfujian, 2023-02-17:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

os version:
ubuntu1~20.04   5.10.0-1057-oem
cpu info: 11th Gen Intel(R) Core(TM) i7-11700 @ 2.50GHz

[root@TVU12F8:/opt/intel/mediasdk/bin]# ./vainfo 
error: can't connect to X server!
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.12 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.2.3 (008fb3b4)
vainfo: Supported profile and entrypoints

It can decode 422 10bit, but fail to decode 422 8bit

 ./sample_decode h265 -i  1676604882563-0.hevc -o test.yuv
1676604882563-0.hevc  is at attachment

[1676604882563-0.zip](https://github.com/intel/media-driver/files/10763100/1676604882563-0.zip)

libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
[ERROR], sts=MFX_ERR_UNSUPPORTED(-3), AllocFrames, m_pmfxDEC->Query failed at /docker/worker/mediasdk/build_dir/repos/MediaSDK/samples/sample_decode/src/pipeline_decode.cpp:1190
[ERROR], sts=MFX_ERR_UNSUPPORTED(-3), Init, AllocFrames failed at /docker/worker/mediasdk/build_dir/repos/MediaSDK/samples/sample_decode/src/pipeline_decode.cpp:593
[ERROR], sts=MFX_ERR_UNSUPPORTED(-3), main, Pipeline.Init failed at /docker/worker/mediasdk/build_dir/repos/MediaSDK/samples/sample_decode/src/sample_decode.cpp:805
[1676604882563-0.zip](https://github.com/intel/media-driver

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

huangli2018, 2023-02-17:
1.	Seems the clip  1676604882563-0.hevc is a wrong clip, its profile is Main10, but format is  422. This is not supported by spec. Main10 profile only support 420.
2.	I also tried this clip on SW decode with 3rd party tool, it can not decode neither.
 Could you please double check the clip or try other 422 8bit hevc clips on your side? Thanks!

intel-mediadev, 2023-02-20:
Auto Created VSMGWL-61663 for further analysis.

Jexu, 2023-02-20:
As description above, this bitstream is invalid when its profile conflicts with surface format. So I will close this issue now and you can re-open it again if still having any other questions.

closed 2023-02-20 by Jexu