intel/media-driver #1616: [Bug]: intel-media.sh causes iHD to always be set as the default driver

llyyr, 2023-02-22:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

This file (https://github.com/intel/media-driver/blob/master/media_driver/cmake/linux/intel-media.sh.in) causes iHD to always be used even when it's not desired, for example on multigpu systems. I'm not sure if this is desired or an oversight, but it's quite inconvenient on a multigpu system and I wonder if it's even necessary on a system with only a single gpu.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery, Playback

### Do you want to contribute a patch to fix the issue?

None

eero-t, 2023-03-02:
Maybe the code for detecting presence of a suitable GPU can be added to it from here: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/gpu_plugin/render-device.sh ?

llyyr, 2023-03-02:
I'm not sure if that'd solve it? It will continue to be detected all the time on a multigpu system. I'm just not sure there's any need to set this variable at all. intel media driver will be used automatically even without setting this variable, and on multigpu systems you really shouldn't be setting this variable.

eero-t, 2023-03-02:
Are you referring to that script being (optionally)  installed to `/etc/profile.d/` (i.e. to env of all processes on that host) by `make install`: https://github.com/intel/media-driver/blob/master/CMakeLists.txt#L90

And if yes, why you enable `INSTALL_DRIVER_SYSCONF` CMake option (it's OFF by default), if you do not want that?

PS. I'm not media-driver developer, just another user.  I've never enabled that option, and this is first time I actually noticed it...

llyyr, 2023-03-02:
oh my bad, it seems my distro turns it on while building...

closed 2023-03-02 by llyyr